z88dk/z88dk #1178: argc corrupt when using implicit int

zx70, 2019-06-05:
Try the sample in the OSCA subdir building it for CP/M.   argc gets a value very big.
Everything is solved by explicitly declaring int (main)

suborb, 2019-06-05:
For compatibility with sdcc main is always a __stdc function. 

I suspect the k&r parsing bit in sccz80 isn’t setting the appropriate flags.

suborb, 2019-06-05:
It wasn't. It should be now!

closed 2019-06-05 by suborb

zx70, 2019-06-12:
I'm not sure it is solved, I still have to add "int" as a prefix to function "main" to make argc/argc work correctly.   Checked with wc.c and comp.c

suborb, 2019-06-12:
I can't reproduce this anymore - is your sccz80 definitely up-to-date?

zx70, 2019-06-12:
I see, I'm probably slightly out of sync.  I'll check it again as soon as "nightly" will be reachable.

suborb, 2019-06-12:
Back up now - it looks like the router crashed this morning.

zx70, 2019-06-12:
problem persists.  edit wc.c and remove "int" before main(.
the built program should now try to open an empty filename in place of the given parameter.

suborb, 2019-06-12:
Got it, thank you. Arrays weren't being setup properly in "implicit int" functions.

zx70, 2019-06-12:
thank you!
I'd be very glad to have also a hint on the readbyte() problem I'm having with the trsdos  🙄

suborb, 2019-06-12:
I think that's my bug too...

zx70, 2019-06-12:
could be, but the cp/m version of the same function works. . it is puzzling me to death

suborb, 2019-06-12:
Yeah, it's really odd. Compiling it, we get this (or something similar, I'm hacking it down at the moment):

```
        push hl  <--- save fastball paramter
        push    bc  <--- make space for struct TRSDOS_FILE *
        ld      hl,6    ;const   <--- Get hold of parameter, this should be +2
        add     hl,sp
        call    l_gint  ;
        pop     bc
        push    hl
```

That ld hl,6 should be ld hl,2. I'm trying to reproduce it in a different file, it's proving tricky though.

suborb, 2019-06-12:
Oh, I'm so, so, very sorry, that's a terrible oversight by me.

Please rename the parameter from "handle" to "fd" and the it'll work as you expect.

zx70, 2019-06-12:
so fast,  thanks!
I presume 'handle' is sort of reserved? 
don't tell me you have the 'fd' exception hardcoded somewhere  😫

suborb, 2019-06-12:
No, it just adjusts the stack locations using the parameter names defined in the prototype (where it's fd) rather than at the function declaration (where the parameter is handle).

So fastcall parameters end up not being moved if there's a naming mismatch.

zx70, 2019-06-12:
ook, that makes sense. :)  ..a warning, in the long term would suffice

commit 651e7834 (2019-06-05): Fix #1178: main() should be __stdc even when declared with k&r