z88dk/z88dk #1187: Implicit Dependency on stdlib.h in ucpp?

QwazyWabbitWOS, 2019-06-08:
Visual Studio (versions tested: 2010, 2017, 2019) spews 'unknown pragma' warnings if <stdlib.h> is not explicitly included in the source modules assert.c, cpp.c, eval.c, lexer.c, macro.c. It looks like not including the header was an oversight as mem.c has it and nhash.c doesn't need it. I can post a PR if desired.

suborb, 2019-06-08:
That's weird, I can't find any `#pragma` within ucpp. Where is it reporting the warning?

QwazyWabbitWOS, 2019-06-08:
It's in the output stream of the VS IDE during the compilation process. The #pragmas are within Microsoft's headers and they're native to their libraries and correctly formatted. 
Sample output:

> 1>cpp.c
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\corecrt_wstdlib.h(362,9): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\corecrt_wstdlib.h(363,9): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\corecrt_wstdlib.h(364,9): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\corecrt_wstdlib.h(373,9): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\stdlib.h(304,9): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\stdlib.h(305,9): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\stdlib.h(339,9): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\stdlib.h(637,9): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\stdlib.h(638,9): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\stdlib.h(639,9): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\stdlib.h(646,9): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\stdlib.h(687,9): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\stdlib.h(688,9): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\stdlib.h(695,9): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\stdlib.h(1098,9): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\stdlib.h(1099,9): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\stdlib.h(1100,9): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\stdlib.h(1109,9): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\stdlib.h(1224,13): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\stdlib.h(1225,13): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\stdlib.h(1238,13): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\stdlib.h(1296,13): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\stdlib.h(1297,13): warning C4068:  unknown pragma
1>C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\ucrt\stdlib.h(1358,13): warning C4068:  unknown pragma

One such line in <stdlib.h> line 304 is:
#pragma warning (push)
#pragma warning (disable:6540)

Which is perfectly normal syntax for it under normal conditions. The IDE identifies the header as an external dependency and apparently includes it automagically. I haven't come up with a small sample project that will reproduce it. I haven't put my finger on the root cause of the inclusion. Seems to be an artifact of the implementation going as far back as VS2010. Microsoft's online help for C4068 is distinctly unhelpful but that's nothing new for them.

The 'fix' is to simply #include <stdlib.h> in the project's source files. The ucpp project is the only one affected.

For example, for cpp.c:

> #include "tune.h"
#include <stdio.h>
#include <string.h>
#include <stdlib.h>
#include <stdarg.h>
#include <setjmp.h>
#include <stddef.h>
#include <limits.h>
#include <time.h>
#include "ucppi.h"
… etc...

suborb, 2019-06-08:
It's still weird! If you can do a PR I'll get it merged in.

closed 2019-06-09 by suborb