z88dk/z88dk #1203: [Xcode] - takes forever and times out

feilipu, 2019-06-18:
I’ve been finding it very difficult to get builds through the Xcode path.
 1. Firstly they take forever, often over 25 minutes vs. about 9 minutes through g++.
 2. Secondly, they’re always failing, probably due to time out.
 3. And this one has now completely stopped.

I don’t know whether this is shouting into a storm, but is this typical for Apple Xcode? Or are we just doing it wrong?

`Successfully installed Capture-Tiny-0.48 (upgraded from 0.23)` took about 5 minutes.
`Building and testing Capture-Tiny-Extended-0.114 ...` is taking over 10 minutes. I’m sitting at 25 minutes now, and nothing z88dk has actually started...

```
Building and testing Test-Warnings-0.026 ... Can't locate TAP/Harness/Env.pm in @INC (you may need to install the TAP::Harness::Env module) (@INC contains: FatPacked::140397833805336=HASH(0x7fb0eb02de18) /Library/Perl/5.18/darwin-thread-multi-2level /Library/Perl/5.18 /Network/Library/Perl/5.18/darwin-thread-multi-2level /Network/Library/Perl/5.18 /Library/Perl/Updates/5.18.2 /System/Library/Perl/5.18/darwin-thread-multi-2level /System/Library/Perl/5.18 /System/Library/Perl/Extras/5.18/darwin-thread-multi-2level /System/Library/Perl/Extras/5.18 .) at /usr/local/bin/cpanm line 458.
The command "cpanm --local-lib=~/perl5 local::lib App::Prove Modern::Perl Capture::Tiny Capture::Tiny::Extended Path::Tiny File::Path Template Template::Plugin::YAML Test::Differences CPU::Z80::Assembler Test::HexDifferences Data::HexDump Object::Tiny::RW Regexp::Common List::Uniq" failed and exited with 1 during .
Your build has been stopped
```

feilipu, 2019-06-18:
Ok new issues...

unknown configuration options.
```
/System/Library/Frameworks/Ruby.framework/Versions/Current/usr/bin/ruby: warning: unknown argument for --enable: `frozen-string-literal'
/System/Library/Frameworks/Ruby.framework/Versions/Current/usr/bin/ruby: warning: unknown argument for --disable: `did_you_mean'
```
moaning about OS x 10.11.
```
Warning: You are using macOS 10.11.
We (and Apple) do not provide support for this old version.
You will encounter build failures with some formulae.
Please create pull requests instead of asking for help on Homebrew's GitHub,
Discourse, Twitter or IRC. You are responsible for resolving any issues you
experience, as you are running this old version.
```
And dead again.
```
Fetching http://www.cpan.org/authors/id/E/ET/ETHER/Test-Warnings-0.026.tar.gz ... OK
Configuring Test-Warnings-0.026 ... OK
Building and testing Test-Warnings-0.026 ... Can't locate TAP/Harness/Env.pm in @INC (you may need to install the TAP::Harness::Env module) (@INC contains: FatPacked::140719142528024=HASH(0x7ffbba80e418) /Library/Perl/5.18/darwin-thread-multi-2level /Library/Perl/5.18 /Network/Library/Perl/5.18/darwin-thread-multi-2level /Network/Library/Perl/5.18 /Library/Perl/Updates/5.18.2 /System/Library/Perl/5.18/darwin-thread-multi-2level /System/Library/Perl/5.18 /System/Library/Perl/Extras/5.18/darwin-thread-multi-2level /System/Library/Perl/Extras/5.18 .) at /usr/local/bin/cpanm line 458.
The command "cpanm --local-lib=~/perl5 local::lib App::Prove Modern::Perl Capture::Tiny Capture::Tiny::Extended Path::Tiny File::Path Template Template::Plugin::YAML Test::Differences CPU::Z80::Assembler Test::HexDifferences Data::HexDump Object::Tiny::RW Regexp::Common List::Uniq" failed and exited with 1 during .
Your build has been stopped.
```

suborb, 2019-06-18:
The Travis OS-X machines are increasingly flaky (See #1145 as well) and I've yet to see a genuine failure that hasn't been picked up earlier by the Linux build.

Most of the slowness if caused by installing all the dependencies needed for the z80asm tests: perl modules, sdcc

Our build environment has a lot of small files, the mac filesystem isn't as bad as NTFS when handling this situation, but it's not as fast as ext4 on the same hardware

The base OS for the Travis OSX is pretty old (El Capitan 2015), hence the scary warnings about unsupported OS.

The Linux machine template is from 2014, but support for it seems to be in slightly better shape and I think they may have local mirrors so dependency installation is quicker.

Given that we don't use any artefacts that are produced by Travis I'm happy with Mac Travis builds being discontinued - I've certainly stopped waiting for the mac build to clear before merging/being happy.

suborb, 2019-06-18:
Reading the docs there’s a newer version of Mac images available. I shall give it a whirl.

closed 2019-06-18 by suborb