z88dk/z88dk #1204: Setting a default output redirection

zx70, 2019-06-18:
TRSDOS does not depend directly to the ROM console service, thus a CMD file will begin printing on the top-left corner rahter than at the current cursor position.
I'm supporting three fake filenames (coherent to the TRSDOS naming convetions) to deal with it:
*DO - display output
*PR - printer output
*KI - keyboard input

The problem is that forcing the console output from within the program (e.g. with freopen) would invalidate any user specified redirection.   
Is there a better workaround than creating  a TRS80 specific variant of crt_command_line ?
Is there a way to understand whether the default stdout was already changed ?

[image]

suborb, 2019-06-18:
There's an `int fchkstd(FILE *)` function that returns whether a file is destined for the console. This is used by the library to determine whether to call `fgetc_cons()`

zx70, 2019-06-18:
if (fchkstd(stdout)==1) freopen("*do","w",stdout);

..does exactly what I was trying to get.

closed 2019-06-19 by zx70

zx70, 2019-06-23:
https://ia801709.us.archive.org/BookReader/BookReaderImages.php?zip=/0/items/MC_C-Language_Compiler_1985_sys_PDF/MC_C-Language_Compiler_1985_Misosys_PDF_jp2.zip&file=MC_C-Language_Compiler_1985_Misosys_PDF_jp2/MC_C-Language_Compiler_1985_Misosys_PDF_0008.jp2&scale=4.436827956989247&rotate=0
..the way the redirection was done at the time :)