z88dk/z88dk #1207: [bug][sdcc][short] I get wrong (just zero) results with sdcc

Fabrizio-Caruso, 2019-06-21:
In my little function that enumerates primitive Pythagorean triples:
https://github.com/Fabrizio-Caruso/Primitive-Pythagorean-Triples

I just get zeros in the result array if I use unsigned short and compile with zsdcc.
I do get good results if I use either any other compiler (including sccz80) or other types for the digits.

The following target produces a good binary:
```
zx_short:
	zcc$(EXEEXT) +zx -O3 -DUSE_SHORT -DTARGET=1000 -DDISPLAY_START=981 PPT.c -lndos -create-app 
```

whereas the following target fails to output the good sequence:
```
zx_short_sdcc:
	zcc$(EXEEXT) +zx -compiler=sdcc -SO3 --max-allocs-per-node200000 -DUSE_SHORT -DTARGET=1000 -DDISPLAY_START=981 PPT.c -lndos -create-app
```

FeliPu, 2019-06-22:
I'm sorry but you tagged the wrong Person..

ped7g, 2019-06-22:
I guess it was meant to be @feilipu  :)

Fabrizio-Caruso, 2019-06-22:
Indeed! But the right person may be @spth

feilipu, 2019-06-23:
Indeed. And if there's a sdcc related issue @spth will ask for a short piece of code that triggers the issue, and to raise a bug request on the sdcc svn at sourceforge.

As an example look at open to move zsdcc build version. There's some bug submit links there.

Fabrizio-Caruso, 2019-06-23:
@spth I have simpliefied the test into just a trivial example that shows the issue (only present in zsdcc):
```
#include <stdio.h>
unsigned short queue[5000];
unsigned short a,b;
unsigned short *head;
unsigned short *tail;
int main(void)
{
    head = queue;
	tail = queue;
    *(tail++) = (unsigned short) 3;
	*(tail++) = (unsigned short) 4;
	a = (unsigned short) *(head++);
	b = (unsigned short) *(head++);
	printf("a: %u\n",a);
	printf("b: %u\n",b);
    return 0;
}
```

Correct result should be the sequence 3,4 on any C compiler.
ZSDCC produces 1027,4. Somehow something goes wrong with the post-increment of the pointers to unsigned short.
I do not see the problem in ZSDCC with unsigned long.

Fabrizio-Caruso, 2019-06-23:
@spth I have also opened an issue in
https://sourceforge.net/p/sdcc/bugs/2917/

Fabrizio-Caruso, 2019-06-23:
Philipp needs to know how Z88DK calls sdcc. Where can I get this information?

suborb, 2019-06-23:
If you supply -v you'll see the command line that's being run. The issue is with the increment of the head/tail:

```
;test.c:10: *(tail++) = (unsigned short) 3;
        ld      hl, (_tail)
        ld      (hl),0x03
        inc     hl
        ld      (hl),0x00
        ld      hl, (_tail)
        inc     hl
        ld      (_tail), hl

suborb, 2019-06-23:
We've got the above PR open to upgrade to r11292 of sdcc. In that revision this issue is now longer observed:

```
;test.c:10: *(tail++) = (unsigned short) 3;
        ld      hl, (_tail)
        ld      (hl),0x03
        inc     hl
        ld      (hl),0x00
        ld      hl, (_tail)
        inc     hl
        inc     hl
        ld      (_tail), hl
```

feilipu, 2019-06-30:
Have a try with sdcc r11311 as per #1149.
It has another issue fixed.

suborb, 2019-07-01:
That version will be in the next nightly so closing this issue since the underlying issue has been solved.

closed 2019-07-01 by suborb