z88dk/z88dk #1211: Secondary VDP on SPC-1000

zx70, 2019-06-28:
I put a simple extra row in the Makefile to build a TMS9918 specific library variant for the SPC-1000.
It relies on a simple lst file (spc9918.lst) to create "gfx2spc1000.lib".
Some of the functions are available already, but in this way the secondary VDP is brought on top of the default settings, i.e. putsprite() or stencil_render() are optimized for the specific hardware.
The "monitor switch" on the emulator must be selected to see the results.

suborb, 2019-06-28:
That was already supported!

zx70, 2019-06-28:
Oops, sorry!
Which was the library name ?  I had the impression that the SPC_1000 was using the slower putsprite3.asm in all the video modes, then calling the appropriate "plot" stuff for the current one.

suborb, 2019-06-28:
It was built into the default spc library. Switching to mode 10+ switched to the vdp. 

There’s a pragma to enable the support. 

It may not have used the optimised versions of course but the low level plotting etc was working.

zx70, 2019-06-28:
Ok, I read your docs correctly, then.
My intention was to have an optimized option in parallel, your clever build process drops the proper objects in the spc100 folder already.  The result will be similar to the one obtained on the MSX target when bsxbios.lib is used, where about 300 bytes are saved with more available functions.

zx70, 2019-06-28:
#1088

zx70, 2019-07-01:
It works, but the new smart functions (plot, point, etc) would bring-in the other video functions.
I'm reverting this change at the moment and try to develop the SPC1000 graphics functions following the same logic.

closed 2019-07-01 by zx70