z88dk/z88dk #1212: (Nightly/g800) Currently sin.c can not be built

fu-sen, 2019-07-01:
I am using the latest Nightly Build on Windows 10.

When I do make in **examples/g800**, it fails in **sin.c**:
[image]

I investigated the cause:
[examples/g800/sin.c](https://github.com/z88dk/z88dk/blob/master/examples/g800/sin.c)
```
    y = (int)(24 * (1.0 - sin(x / 144.0 * 3.14 * 8)));
```
It seems that there was a problem with floating point calculations because "x" is at the beginning of the expression.
I rearranged the formula and added a decimal. As a result, it worked:
```
    y = (int)(24 * (1.0 - sin(3.14 * 8.0 * x / 144.0)));
```
I could send a pull request, but that would be an easy fix.

suborb, 2019-07-01:
@fu-sen thanks for the heads up. I think it's a problem with not promoting the integer LHS to double when the RHS is already a double.

I'll get in a fix soon

closed 2019-07-01 by suborb

fu-sen, 2019-07-02:
Examples/g800/make works fine with the latest build updated:
[image]