z88dk/z88dk #1213: (Nightly/msx) use of sprites in the glx library makes the display strange.

fu-sen, 2019-07-01:
When I run [examples/msx](https://github.com/z88dk/z88dk/tree/master/examples/msx) with the latest Nightly Build,
I get display strange on some sources:
[image]
[image]
This seems to be showing SCREEN 3 (multi color mode) because it is a complete block and sprite.
Or it is completely black and no text is displayed.

It is the following source that gets these indications:
- ex4.c
- ex6.c
- ex7.c
- ex9.c
- ex11.c
- showlib3d.c

Many of them use sprites. It seems that 3D library will be added there.

fu-sen, 2019-07-29:
I was searching for this cause. It is failing with **set_mangled_mode()**.
It is necessary to change VDP to SCREEN 2 after changing to SCREEN 1, which **requires a BIOS call.**
But you changed VDP directly to change the screen mode. It seems to be lost.

fu-sen, 2019-07-29:
In the sample using "set_mode (mode_2);", some samples work, but ex9.c does not work.
So it seems that sprites also have problems.

zx70, 2019-10-01:
your theory about the bios requirements could be partially verified with asimple test:  what happens if you add -lmsxbios to your compile command chain? 
please try also by running the makefile

fu-sen, 2019-10-07:
You forgot past work. It is included in the Makefile:
[z88dk/examples/msx/Makefile](https://github.com/z88dk/z88dk/blob/master/examples/msx/Makefile):
```
EX8.COM: ex8.c
	$(CC) +msx -vn -subtype=msxdos -create-app -oex8 ex8.c -lmsxbios

ADVA.COM: ../console/adv_a.c
	$(CC) +msx -vn -subtype=msxdos ../console/adv_a.c -I../console -DCOMPUTE_SCORE -DPICS -DDISK -create-app -clib=ansi -oadva.bin

ex6.cas: ex6.c
	$(CC) $(CFLAGS) -oex6.bin ex6.c -lmsxbios

ex7.cas: ex7.c
	$(CC) $(CFLAGS) -oex7.bin ex7.c -lmsxbios

showlib3d.cas: showlib3d.c
	$(CC) $(CFLAGS) -oshowlib3d.bin showlib3d.c -lmsxbios

ex11.cas: ex11.c
	$(CC) $(CFLAGS) -oex11.bin ex11.c -lmsxbios
```

zx70, 2019-10-07:
I forget many things :)
Is it still broken? I know there are still limits on the SVI ports but the MSX part looks ok to me.

fu-sen, 2019-10-07:
Unfortunately, this issue is ongoing in the latest build (20191007).
Execution result of ex6.cas using blueMSX:
[image]

suborb, 2019-10-11:
I've just merged #1300 which puts the sprite address back to where they are the MSX. The sprite demos now seem to work. 

Additionally, msx_set_mangled_mode() is now restored.

If there's still issues, please reopen.

closed 2019-10-11 by suborb

fu-sen, 2019-10-12:
suborb Great fix!

Execute ex7 with blueMSX:
[image]