z88dk/z88dk #1214: (libsrc/video/tms9918/gen_set_mode.asm) The modes of SCREEN 0 and 1 are reversed

fu-sen, 2019-07-01:
When I looked at the source of [libsrc/video/tms9918/gen_set_mode.asm](https://github.com/z88dk/z88dk/blob/master/libsrc/video/tms9918/gen_set_mode.asm), I noticed an error:
```
; Mode 0: 32x24
```
```
; Switch 2 Video Mode n. 1
; 40x24
```

Mode 0 and Mode 1 appear to be reversed:
https://en.wikipedia.org/wiki/Texas_Instruments_TMS9918#Screen_modes

For example, although MSX's SCREEN 1 can use sprites and has WIDTH 32,
but the source is not defined in Mode 1:
```
    ld    a,$76           ; Unused (sprites inactive)
    call  VDPreg_Write    ; reg5  -  SPRITE ATTR. TAB.
    
    ld    a,$07           ; Unused (sprites inactive)
    call  VDPreg_Write    ; reg6  -  SPRITE PATTERN GEN. TAB.
```

Is it reversed with other considerations? If so at least MSX users will be confused!

suborb, 2019-07-01:
The VDP code is used by multiple platforms - when I did a reorg I tried to align the screen modes with the VDP documentation rather than the MSX convention - I was getting in a bit of a tangle with them.

The offsets are different to standard MSX as well, but it should all be documented on the wiki: https://github.com/z88dk/z88dk/wiki/Classic-TMS9918

The core library uses the hardware directly and as such should be the same across all targets. For the MSX and SVI there's an additional `-lmsxbios` library which calls via the firmware and by the looks of things does (confusingly) have the modes in MSX conventional order.

I'm sure there are a few issues with the code - I've not had a chance to test the sprites myself, but I think @zx70 was doing some work around here lately?

zx70, 2019-07-01:
I focused mostly on mode 2 (6144 bytes graphics and 6144 bytes attributes).
the msxbios libraries, as Dom said, should work as you expected.   svibios in turn could 'confuse' the SVI users because of the existence of the 32x24 text mode  which I inserted to increase the portability.
I understand now that we have a misalignment between the library directly accessing the vdp and the bios one..  but I have no suggestion for it

fu-sen, 2019-07-01:
I understood. But that would require a note on the wiki.

I was aware of the difference in addresses (Sprite Attribute=0x3b00, MSX usually uses 0x1b00). z88dk seems to be able to control it. but there is a possibility that problems may occur.
I will test in text mode. This bug may issue another issue separately. It can be linked here.

zx70, 2019-07-02:
https://github.com/z88dk/z88dk/blob/master/examples/msx/vpoke.c
I'm a bit out of topic, but this example should work on both svi and msx, probably only if msxbios or svibios are linked in.

fu-sen, 2019-07-02:
> https://github.com/z88dk/z88dk/blob/master/examples/msx/vpoke.c
> I'm a bit out of topic, but this example should work on both svi and msx, probably only if msxbios or svibios are linked in.

I tried immediately:
- zcc +msx -lndos -create-app vpoke.c (It is in the comment) : problem with the display
- zcc +msx -lndos **-lmsxbios** -create-app vpoke.c : worked!

vpoke.c and make will need this rewrite.

fu-sen, 2019-07-02:
The following code works fine with -lmsxbios:
```
#include <msx/gfx.h>

main(){
    set_mode(mode_1);
}
```
The display of SCREEN 1 could be resolved with the addition of -lmsxbios.

But when changing to **mode_0**, this is not working:
[image]
SCREEN 0 has no border color. This appears to be maintaining SCREEN 1-3. This is another problem.

The problem is changing so I can change the title.

suborb, 2019-10-11:
I've fixed up the generic `msx_set_color()` so it should have the same behaviour as the firmware call. These values are also preserved over screen mode change. If this doesn't solve the problems then please reopen!

closed 2019-10-11 by suborb

fu-sen, 2019-10-12:
@sudorb, good job!

I have confirmed that this problem has been solved:
```
#include <msx/gfx.h>

main(){
    set_mode(mode_0);
}
```
```
zcc +msx -create-app -lmsxbios -subtype=disk -bn screen0 screen0.c
```
[image]

```
#include <msx/gfx.h>

main(){
    set_mode(mode_0);
}
```
```
zcc +msx -create-app -lmsxbios -subtype=disk -bn screen1 screen1.c
```
[image]