z88dk/z88dk #1215: (lib/config/g800.cfg) Using g800_clib when CLIB=g850b

fu-sen, 2019-07-01:
The title is as follows:
[lib/config/g800.cfg](https://github.com/z88dk/z88dk/blob/master/lib/config/g800.cfg)
```
CLIB      g850b   -pragma-need=ansiterminal -D__CONIO_VT100 -pragma-export:ansirows=6  -lg800_clib -lndos
```
There is "-l**g800**_clib".
From this, for example, getmaxy() gets 31(=4*8-1).
It gets disappointing display on some graphics of PC-G850.

[examples/g800/mandel.c](https://github.com/z88dk/z88dk/blob/master/examples/g800/mandel.c):
[image]

Maybe CLIB=g850 also got the output which is not supposed:
```
CLIB      g850   -pragma-need=ansiterminal -D__CONIO_VT100 -pragma-export:ansirows=8 -pragma-export:ansicolumns=36 -pragma-redirect:ansifont=_font_3x5_850 -pragma-define:ansifont_is_packed=0 -lg850_clib -lndos
```

fu-sen, 2019-07-01:
I examined in detail. As a result, this is 3x5 in the PC-G850 series, so it seems to work correctly:

> Maybe CLIB=g850 also got the output which is not supposed:
> 
> ```
> CLIB      g850   -pragma-need=ansiterminal -D__CONIO_VT100 -pragma-export:ansirows=8 -pragma-export:ansicolumns=36 -pragma-redirect:ansifont=_font_3x5_850 -pragma-define:ansifont_is_packed=0 -lg850_clib -lndos
> ```

On the other hand, there is a problem in that "4 rows text" is included in the corresponding page of wiki.
This should be at "6 rows text":

https://github.com/z88dk/z88dk/wiki/Platform---Sharp-PC#command-line

zx70, 2019-07-03:
I hope I fully understood you report. 
Please revise it and close the issue if the problem is now solved.   I see you opened a parallel issue for the documentation gaps.

fu-sen, 2019-07-03:
Thanks for the quick source improvement. After reflecting on Tomorrow's Nightly, I will confirm it.

fu-sen, 2019-07-04:
I was able to confirm the improvement of this issue.
This is mandel.ihx which I got by making with latest Nightly Build:
[image]
I can close it:

zx70, 2019-07-04:
I'm afraid that this change is not coherent with the text modes, I need to mix the library sources properly.

fu-sen, 2019-07-04:
Oh, I forgot that confirmation. Now that text is lost.
[image]

zx70, 2019-07-04:
New approach, now I'm attaching only a part of the g850 library (graphics output and keyboard input) to the g800 one.
In this way the 6 rows text mode should be kept but the graphics code will be replaced.

fu-sen, 2019-07-05:
My PC has been updated to Night Build 20190705.
Currently, make is displayed correctly because mandel is built with CLIB = g850:
[image]

Problems will occur if CLIB=g850b. Therefore this issue is ongoing:
[image]

The text display at CLIB=g850b has recovered:
```
#include <stdio.h>

void main() {
  printf(12);

  for (int i=1; i<=6; i++) {
    printf("%c[%u;%uH", 27, i, i);
    printf("test");
  }

  while (!getk());
}
```
 zcc +g800 -create-app -clib=g850b -bn text.ihx text.c
[image]

zx70, 2019-07-05:
Sorry for the mess  :construction:
The library linking priority is opposite on the CFG file respect to what happens when when it is user supplied.
You can wait for the next nightly build or simply add also "-lg850b" to your zcc command line.

fu-sen, 2019-07-06:
My PC has been updated to Night Build 20190706.
```
zcc +g800 -create-app -clib=g850b -bn text.ihx text.c
g800 text.ihx 100
```
[image]
```
zcc +g800 -lm -create-app -clib=g850b -o mandel.ihx mandel.c
g800 mandel.ihx 100
```
[image]
it is perfect! I will test a little more but this issue will close soon.

closed 2019-07-10 by zx70

zx70, 2019-07-17:
G850 graphics and console handling are now fairly complete,  a good part of the generic demos should work too

fu-sen, 2019-07-17:
I understood. Thank you for the source improvement for a while.
I will test again several days.

fu-sen, 2019-07-19:
I tested with the latest Nightly Build (20190720)
examples/g800 works fine..
But I confirmed the new problem in the text output.
This may have existed before and will be a separate issue.