z88dk/z88dk #1217: (Wiki/g800) Description of "CLIB=g850" is added to "4 rows text"

fu-sen, 2019-07-02:
there is a problem in that "4 rows text" is included in the corresponding page of wiki.
This should be at "6 rows text":

https://github.com/z88dk/z88dk/wiki/Platform---Sharp-PC#command-line

_Originally posted by @fu-sen in https://github.com/z88dk/z88dk/issues/1215#issuecomment-507293197_

zx70, 2019-07-03:
I corrected the documentation basing on the current configuration options.
I haven't repeated the checks but the text resolution should be selectable between 4, 6 or 8 rows.

fu-sen, 2019-07-03:
I try it out and understand:

https://poke-com.jimdo.com/2019/07/02/2019-07-02-z88dk-clib-g850b-%E3%81%A8-clib-g850-%E3%81%AE%E9%81%95%E3%81%84/ (Japanese)

fu-sen, 2019-07-03:
Since this issue is being resolved, I can close it.

closed 2019-07-03 by fu-sen

zx70, 2019-07-03:
Wonderful,  thanks for pointing out we forgot to update the wiki.
There is still room for improvements (e.g. to keep a buffer for text attributes), but I preferred to simplify and save memory.