z88dk/z88dk #1218: Jupiter ACE printing to screen crash

benkid77, 2019-07-02:
Hello,

I compile the below C program with 
`zcc +ace -create-app -o printbug printbug.c`
and run it with
`0 0 bload printbug`

After a short while, the program appears to overwrite RAM outside of the screen area, including the character definition RAM, eventually crashing the emulated Jupiter ACE.

However, if the code is changed to `n=31` or `n=33` then the program does not crash, but runs indefinitely with no problems.

So the issue appears to be related to perhaps printing a carriage return on the very end of any row.

```
#include <stdio.h>

int main() 
{
int i,n=32;

for(;;)
	{
	for(i=0;i<n;i++)
		putchar('a');
      	putchar('\n');
	}
}
```

zx70, 2019-07-04:
The code dealing with the console output is very simple, it just turns the characters to be printed to the ROM (entry $3ff).   I see a double "linefeed" being printed, I wonder if it is part of the problem (control codes?).
While we look for a workaround, I suppose you can switch to the full VT emulation (-clib=ansi).

benkid77, 2019-07-04:
Thank you for the reply and for looking into this.

It's OK, I have worked around the problem by using a small wrapper function which uses the ROM "PRINT Restart" routine (rst 0x08). This seems to be working absolutely fine in my program, so I am not waiting on a solution.

Glad I found the issue though so at least I could report it.

zx70, 2019-07-04:
Good hint !  I'll try to move to RST8 also in the library.

zx70, 2019-07-04:
I'm experimenting the same problem with rst8, would you mind to share your workaround ?

zx70, 2019-07-04:
Never mind, I think it is a 'bug' in the ROM needing a workaround, instruction at L0416 will **first** increment the text position, **then** check if the rightmost column has been reached :

(...)
        LD      (HL),A                  ; else insert the character.

        INC     HL                      ; point to next position.
        JR      L041C                   ; forward

; ---

; a carriage return

L0416:  INC     HL                      ; increment screen address.
        LD      A,L                     ; fetch low byte of address and mask.
        AND     $1F                     ; a zero result indicates a line skip.
        JR      NZ,L0416                ; loop until a new line of 32 columns
                                        ; is started.

; both paths converge.

L041C:  LD      ($3C1C),HL              ; update SCRPOS
(...)

benkid77, 2019-07-04:
Sure, if it's any help at all, the bit where I convert '\n'to '\r' is tailored specifically to my application.

```
void my_putchar(char c)
{
#asm
        pop bc		//pop return address
        pop de		//pop argument
        ld a,e		//low byte of argument (char c) into a
        cp '\n'		//convert any newlines to carriage returns as used by the ROM

	jr nz,L_NOTCR
        ld a,'\r'

L_NOTCR:

	push de		//push argument back onto stack
        push bc		//push return address

        rst 0x08	//call the PRINT restart
#endasm
}
```

benkid77, 2019-07-04:
> Never mind, I think it is a 'bug' in the ROM needing a workaround, instruction at L0416 will **first** increment the text position, **then** check if the rightmost column has been reached :
> 
Ah OK, thanks for the further investigations.

zx70, 2019-08-27:
I got an unexpected clue building mazezam.
With clib=ansi  on the j.ace it is necessary to pass  explicitly the column size. 
It still does not explain everything,  but it is something

zx70, 2019-08-27:
ok not sure,  but this combination fixes a couple of problems on the console for the mazezam game:

    -clib=ansi -pragma-define:ansicolumns=32

zx70, 2019-08-28:
I finally discovered that I was looking in the wrong place.
The default console output is now delegated to the new generic conout library, the native one is a third variant, which can be enabled with "-pragma-redirect:fputc_cons=fputc_cons_native"

suborb, 2019-10-11:
I believe I've now fixed this - certainly the code snippet above works doesn't crash.

commit 457b45cf (2019-10-12): (gencon) Fix #1218 in a way that doesn't break everything
closed 2019-10-11 by suborb