z88dk/z88dk #1219: z80asm documentation

obiwanjacobi, 2019-07-03:
I am looking for the detailed syntax docs (not cmd line options) on the z80asm assembler/linker.
I seem to remember there being a (probably old) text document, but I cannot find it anywhere.

Is there an easy way to have an overview of the allowed syntax in the assembler?
Is there a sample source file that demonstrates all options?

pauloscustodio, 2019-07-03:
There is a probably out-of-date document in https://github.com/z88dk/z88dk/blob/master/doc/z80asm.html

obiwanjacobi, 2019-07-03:
LIB, XDEF and XREF are obsolete right?

pauloscustodio, 2019-07-03:
Yes, they have been superseeded by EXTERN and PUBLIC, but they are still recognized.

There is also GLOBAL declares a symbol either as PUBLIC, if it exists in the current module, or as EXTERN otherwise.

obiwanjacobi, 2019-07-04:
I don't see SECTION. Any other new stuff that's been added and is not in the docs?

pauloscustodio, 2019-07-04:
SECTION name: start a new named section or continue a previous section
with the same name
ORG at the start of a SECTION sets the section to absolute address,
causing a new binary to be generated by the linker
SECTION without an ORG follows the previous section in the same binary file
ORG -1 tells the linker to split this section to a separate binary file

PHASE addr: assemble the following code as if it was located at the
given addr - it will be relocated in run-time
UNPHASE: ends phased code

These are from the top of my head... there may be more.

I'll keep this issue open as a reminder to update the documentation.

feilipu, 2019-07-04:
Another one is `ALIGN` which allows you to space code to specific location, set to a specific starting address, often a page alignment.

There is some usage of `SECTION` to put code into a specific grouping for the linker, `PHASE` to tell the assembler that it needs to have a different origin than the `SECTION` it is being put into, and `ALIGN` to space the code to fit on specific pages, [here](https://github.com/feilipu/yaz180/blob/master/yabios/asm_common1_data.asm), and [here](https://github.com/feilipu/yaz180/blob/master/yabios/asm_common1_driver.asm).

obiwanjacobi, 2019-07-04:
Thanks guys, good info.

Can defb, defw and friends be shorted to 'db', 'dw'? 
In the html I see a sample for DEFVARS that uses 'ds.b 1' and 'ds.b 6' - is that still supported or only valid inside a DEFVARS block (not a general alias for defs)?

feilipu, 2019-07-04:
Just to add `DEFQ` as undocumented friend to `DEFW` and `DEFB`. Used [here](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/math32/z80/h32_coeff_atan.asm).

obiwanjacobi, 2019-07-04:
What expression does DEFQ accept? Does it always have to be hex (32) or can it be decimal. Can you use symbols (defc's), something like `symbol1 + 0x10000000` ?

pauloscustodio, 2019-07-04:
DEFQ reserves space for a 32-bit value. All the DEFx instructions accept expressions to be computed at assemble or link time.

obiwanjacobi, 2019-07-05:
> Can defb, defw and friends be shorted to 'db', 'dw'?
> In the html I see a sample for DEFVARS that uses 'ds.b 1' and 'ds.b 6' - is that still supported or only valid inside a DEFVARS block (not a general alias for defs)?

And this ^^^?

pauloscustodio, 2019-07-05:
> Can defb, defw and friends be shorted to 'db', 'dw'?
They cannot, but can with a very small change - I'll add it to the feature requests (#1221).

pauloscustodio, 2019-07-05:
> In the html I see a sample for DEFVARS that uses 'ds.b 1' and 'ds.b 6' - is that still supported or only valid inside a DEFVARS block (not a general alias for defs)?

Yes, DS.B, DS.W, DS.P and DS.L are only valid inside DEFVARS. This is a way to define data structures.

obiwanjacobi, 2019-07-06:
It was not my intention to request for new features. Nonetheless I think it would make a good addition in line with the syntax of other assemblers.

I'm just trying to write a Z80 (antlr) grammer for the z88dk z80asm...
https://github.com/obiwanjacobi/Zingularity/blob/master/Source/Language%20Server/server/src/z80asm/z80asm.g4
Still to be tested and implemented further...

feilipu, 2019-09-30:
Seems complete. Please reopen if not.

closed 2019-09-30 by feilipu