z88dk/z88dk #1220: (msx) The value of msx_type such as MSX2 is incorrect and MSX2+ / turbo R correspondence

fu-sen, 2019-07-05:
I discovered that the value of msx_type is strange except for MSX1 (mostly MSX2):
```
#include <stdio.h>
#include <msx.h>

main () {
   printf("%u", msx_type());
}
```
```
 zcc +msx -lmsxbios -create-app -bn msxtype msxtype.c
```
MSX1 (correct):
[image]

MSX2 (The correct value is 2):
[image]

I found the cause:
https://github.com/z88dk/z88dk/blob/master/libsrc/target/msx/msxbios/msx_type.asm:
```
ld	h,3
```
Therefore, 3(H)✕256+2(L=MSX2)=770

Consider MSX2 + and MSX turbo R together.
MSX, 2, 2+ and turbo R have their value at $002d:
http://map.grauw.nl/resources/msxsystemvars.php

zx70, 2019-07-05:
Thankyou for spotting it so precisely and for the further information !
At the moment I'm avoiding to extend this function, though.   If you think it can be useful I'd rather provide a pointer definition in the header file, or you can just define it yourself:
extern unsigned char  MSX2_SUBVER @0x002d;

There are at least two other ways to make a similar definition (probably this is the worse one) but I'm addicted to it   :)

fu-sen, 2019-07-05:
Perhaps I think it's used internally, so I understand that suggestion.
It is a simple and ideal implementation.

zx70, 2019-07-05:
My major concern is to avoid to ungroup the MSX2 machines family in msx_type().   I'm adding MSX2_SUBTYPE in the header file.

fu-sen, 2019-07-05:
The implementation is fulfilled in response to this issue. I will test later.

fu-sen, 2019-07-06:
My PC updated Nightly Build 20190706 and tried the following source "msxtype.c":
```
#include <stdio.h>
#include <msx.h>

main () {
   printf("msx_type()  =%u\n", msx_type());
   printf("MSX2_SUBTYPE=%u\n", MSX2_SUBTYPE);
}
```
MSX1 is mainly a cassette. On the other hand, MSXturboR has the cassette removed.
Created cassette tape image msxtype.cas file and disk BLOAD format msxtype.msx file:
```
zcc +msx -lmsxbios -create-app -bn msxtype msxtype.c
zcc +msx -lmsxbios -create-app -subtype=disk -bn msxtype msxtype.c
```
MSX1:
[image]
MSX2:
[image]
MSX2+:
[image]
MSX turbo R:
[image]

MSX is perfect!
I do not have the environment of SVI-318 and can not test, but this issue may be closed.

zx70, 2019-07-06:
Thanks for testing

closed 2019-07-06 by zx70