z88dk/z88dk #1222: (pc6001/graphics.h) todraw is drawn in the upper left of the screen

fu-sen, 2019-07-07:
I have tried z88dk on some platforms.
I found a strange behavior in the mono graphic of PC-6001 in it:
```
#include <stdio.h>
#include <graphics.h>

void main() {
  clg();

  plot(10, 10);
  drawto(30, 30);

  while (!getk());
}
```
zcc +pc6001 -create-app -bn grptest grptest.c
[image]
When drawto is set to plot, it draws normally.

Some other mono graphics commands will do strange things.

suborb, 2019-07-07:
Looks like I forgot to update the last plotted coordinate variable when I added the extra modes.

There may be a few more ports that have the same problem even after this fix, if you find them let me know...

fu-sen, 2019-07-07:
That is the reason, Because I am using Kokan BASIC(互換 BASIC=ごかんBASIC=compatible BASIC)
in consideration of copyright:
http://000.la.coocan.jp/p6/basic.html

fu-sen, 2019-07-07:
I confirmed the code change. I will test after reflecting the Nightly Build tomorrow...

zx70, 2019-07-07:
please ignore my last commit, I reverted the change already

fu-sen, 2019-07-08:
My PC has been updated to Night Build 20190708.

The following results are obtained for grptest.c published in this issue. (correct):
[image]

I was making a sine curve with mono graphic. The PC-6001 now works properly:
https://gist.github.com/fu-sen/4e079229d43a0aa6c9d40734c35fb2ab
[image]

I have tried other platforms but I have not found the same problem. Therefore this issue is closed.

fu-sen, 2019-07-09:
Unfortunately I reopened this issue.
After that I was able to try other modes (mode 1 and 2) with PC-6001ml2 compatible BASIC,

```
#include <graphics.h>
#include <math.h>
#include <stdio.h>
#include <sys/ioctl.h>

void main() {
  int mode = 1;
  int x, y;

  console_ioctl(IOCTL_GENCON_SET_MODE, &mode);

  clg();

  for (x=0; x<=getmaxx(); x+=2) {
    plot(x, getmaxy() / 2);
  }

  for (x=0; x<=getmaxx(); x++) {
    y = (int)((getmaxy() / 2) * (1.0 - sin(M_PI * 2.0 * x / getmaxx())));
    if (x==0){
      plot(x, y);
    } else {
      drawto(x, y);
    }
  }

  while (getk() < 1){}
}
```

mode 0 (correct):
[image]
mode 1:
[image]
mode 2:
[image]

I changed it to plot only without using drawto for trial.
```
  for (x=0; x<=getmaxx(); x++) {
    y = (int)((getmaxy() / 2) * (1.0 - sin(M_PI * 2.0 * x / getmaxx())));
      plot(x, y);
  }
```
It works:
[image]
As a result, when in Mode 1 and 2, this problem is continuing.

closed 2019-07-09 by suborb

fu-sen, 2019-07-11:
I was able to confirm the solution of the problem.

Mode 1:
[image]

Mode 2:
[image]

This will advance the development for PC-6001 using z88dk Thank you!

commit d3908de1 (2019-07-09): Fix #1222: Set the static screensize to the maximum resolution