z88dk/z88dk #1223: (classic) BBC BASIC floating point library

suborb, 2019-07-08:
BBC BASIC for z80 is now under the MIT licence. Source can be found here: https://github.com/davidgiven/cpmish/tree/master/third_party/bbcbasic

This is the same maths code as used on the z88 so would be interesting to benchmark alongside genmath, math48.

feilipu, 2019-07-08:
It looks like an [interesting library](https://github.com/davidgiven/cpmish/blob/master/third_party/bbcbasic/fpp.z80). It is a [40 bit solution](https://github.com/davidgiven/cpmish/blob/master/third_party/bbcbasic/fpp.z80#L8), half way between math48 and IEEE.

They use [polynomials](https://github.com/davidgiven/cpmish/blob/master/third_party/bbcbasic/fpp.z80#L2011) extensively, and with a 32 bit mantissa, the same algorithms would work on math32 in its expanded mantissa internal calculations.

Just, don't ask it to do fast [integer division](https://github.com/davidgiven/cpmish/blob/master/third_party/bbcbasic/fpp.z80#L331). I think someone cheated, probably to save ROM space. :smile:

suborb, 2019-07-09:
The printing isn't quite right, but n-body comes out to: 1_655_772_821 so between math32 and mbf32.

Not too shabby really.

suborb, 2019-07-12:
Merged as #1225

closed 2019-07-12 by suborb