z88dk/z88dk #1224: math32 - resolve inaccuracy in benchmarks

feilipu, 2019-07-10:
Both [benchmarks](https://github.com/z88dk/z88dk/wiki/Classic--Maths-Libraries#benchmarks) used in the Classic Math Library wiki are simply using the 4 basic arithmetic operations. There are no derived operations needed.
- [spectral-norm](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/spectral-norm/z88dk-new/spectral-norm.c)
- [n-body](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/n-body/z88dk-new/n-body.c)

The math32 library produces accurate results for the 4 basic operations, yet the results for the benchmarks are incorrect. That is very peculiar, and needs to be resolved.

feilipu, 2019-07-10:
Just a thought perhaps the `double` declaration is not working right? Hence `-Ddouble=float`.

suborb, 2019-07-10:
Both compilers map float/double to their supported format. 

We get the same result with sdcc so it’s definitely libraries - it could be an issue with ftoa so it’s worthwhile grabbing the bytes and checking what is actually being calculated.

feilipu, 2019-07-10:
I can't find an example of the `dload` function made for 4 byte floats anywhere. `dload` is used by `dtoa` (mapped to `ftoa`).

The only examples relate to `math48` loading 6 bytes into registers of `AC'`.
Perhaps that is a problem?

Investigate why the SDCC math libraries then work?
Probably because they don't touch z88dk classic or newlib, therefore their 4 byte float conversion works fine.

suborb, 2019-07-10:
I was referring to classic which has its own `ftoa()` implementation in libsrc/math/cimpl - the same code will be called by sdcc and sccz80.

That code was originally written for genmath and has had to have  a few tweaks for the newer maths libraries - it should still work correctly.

The 32 bit maths libraries use the `l_f32` prefix, hence for newlib `dtoa` needs to be part of the maths library rather than the core platform library

I suspect I might have to re-run the benchmarks, I don't things are quite as bad as they seem:

```
math32/n-body:
be 2d 22 11
-0.169075264
be 2d 24 e7
-0.169086089
```

The numbers are closer to the "correct" values, and the printed numbers match those printed for the same bytes on a different machine.

feilipu, 2019-07-10:
Ok. Then, I'll stand down for a while.

I'd consider that anything below the [7th decimal place](https://en.wikipedia.org/wiki/Floating-point_arithmetic#Internal_representation) is noise from the 24-bit mantissa point of view. 4 Byte IEEE 754 simply isn't as accurate as the 5 Byte and 6 Byte options. It's where the 6th and 7th decimal digit start to be incorrect that I'm really concerned.

Are your benchmark runs the same as noted in the[ notes for n-body](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/spectral-norm/z88dk-classic), adjusted with the `-lmath32` calling incantation?
```
sccz80/classic
zcc +test -vn -DSTATIC -DTIMER -D__Z88DK -O2 spectral-norm.c -o spectral-norm.bin -lm -lndos -m
```
The [notes suggest that spectral-norm](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/n-body/z88dk-classic) fails with sccz80. But that can't be the current situation. Is there an incantation that works, that I can test against?

I'm interested to play with the n-body benchmark, and properly calling [one line](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/n-body/z88dk-classic/n-body.c#L72) with the `insqrtf()` function would make a huge difference in the score.

suborb, 2019-07-10:
I'm compiling and running as follows:
 
```
zcc +test -DPRINTF spectral-norm.c --math32
z88dk-ticks -w 60 a.bin
```

And likewise with `n-body.c`

I'm not sure why it previously didn't compile, possibly something to do with arrays? Regardless, it now does work so I'm not going to worry.

feilipu, 2019-07-11:
As you point out, the answers seem to be actually pretty correct (within the 7 significant digit limit). :+1:
```sh
zcc +test -DPRINTF n-body.c --math32
z88dk-ticks -w 60 a.bin
-0.169075264
-0.169086089
Ticks: 1400110112
```
I [committed the `math32` _intrinsic_ functions](https://github.com/z88dk/z88dk/commit/4f3e250304bac40f88f6ada6e6d3e5c5938052b9) to the classic library, which allows us to add a configuration...
```C
#ifdef __MATH_MATH32
      inv_distance = invsqrt(sqr(dx) + sqr(dy) + sqr(dz));
#else
      inv_distance = 1.0/sqrt(dx * dx + dy * dy + dz * dz);
#endif
```
And we get about a __25% improvement__.
Most of this gain is created by directly using the `invsqrt()` function.
Effectively `y=invsqrt(x)`, instead of actually calculating `y=l_f32_inv(x*invsqrt(x))`

The `sqr()` function only adds about 3% additional gain.

```sh
zcc +test -DPRINTF n-body.c --math32 -D__MATH_MATH32
z88dk-ticks -w 60 a.bin
-0.169075264
-0.169086709
Ticks: 1040434814
```
Is this cheating? Well the formula is still generalised and written in C, so I'd suggest that it is more of a configuration than a cheat. :smile:

feilipu, 2019-07-12:
I did the benchmarks again for z80, but I can't reproduce the z180 nor zxn benchmarks.
I'm not sure of the correct incantation. Sorry.

closed 2019-07-12 by feilipu

suborb, 2019-07-12:
@feilipu:

```
zcc +test -DPRINTF n-body.c --math32 -clib=z180 -lmath32_z180
z88dk-ticks -w 60 -mz180 a.bin
```

and

```
zcc +test -DPRINTF n-body.c --math32  -clib=zxn -lmath32_zxn
z88dk-ticks -w 60 -mz80-zxn a.bin
```

feilipu, 2019-07-12:
Thanks, I spent nearly an hour going through all the combinations last night (and I thought I knew what I was doing). I guess that happens when you don't use the platform enough. :confused:

#1227.

For later repeatablity.

```
zcc +test -v -Cc-mz180 -Cc-fp-mode=ieee -lmath32_z180 -custom-copt-rules=~/Z80/z88dk/lib/z180_rules.1 -startuplib=z180_crt0 -pragma-define:CLIB_32BIT_FLOATS=1 -DPRINTF -D__MATH_MATH32 n-body.c
z88dk-ticks -w 60 -mz180 a.bin
```

```
zcc +test -v -Cc-mz80-zxn -Cc-fp-mode=ieee -lmath32_zxn -custom-copt-rules=~/Z80/z88dk/lib/zxn_rules.1 -clib=zxn -pragma-define:CLIB_32BIT_FLOATS=1 -DPRINTF -D__MATH_MATH32 n-body.c
z88dk-ticks -w 60 -mz80-zxn a.bin
```