z88dk/z88dk #1226: 8080 support

suborb, 2019-07-12:
I keep coming back to this in my head. 

I think we can probably do something here that’s not too intrusive. 

* Cleanup stack by moving hl into bc
* Disable shift optimisations
* Enforce 32 bit maths - mbf32 is 8080 code
* Replace all 16 bit sbc with 8 bit equivalent
* Replace ld de,() with ex de,hl pair + hl loss
* Steal long crt0 from gbdk?
* Assess opt rules (mark with cpu flag)
* Disable inline functions for 8080
* Review c stdlib
* Revive old 8080 stdio
* Disable extended opcodes in ticks and error so we catch stuff. 

The main target would be cpm of course.

feilipu, 2019-07-13:
Another 8008 and 8080 32-bit math option could be built too, the [Lawrence Livermore (National) Labs Floating Point Library](https://github.com/feilipu/LLL-Floating-Point), circa 1973.

Integrating this library into z88dk is on my list (admittedly sidelined until we have 32-bit math support in newlib), because of its commonality with Am9511A. I'm not 100% certain, but I'd guess that the Am9511A was built to replicate the LLL floating point format.

suborb, 2019-07-13:
I had a look at that last night - I couldn’t see the trig functions - did I miss them?

feilipu, 2019-07-13:
No. There's nothing except the 4 arithmetic functions, and some helper functions to convert fixed to float and back, in the library. Pretty basic. But probably used to design ICBM flight paths, amongst other things.

This is the [original 8080 code](https://github.com/feilipu/LLL-Floating-Point/blob/master/80_lllf.asm), before conversion to Zilog mnemonics, from [Herb's stuff](http://www.retrotechnology.com/herbs_stuff/float.html).

suborb, 2019-07-15:
This is a little bit of an ordeal but the basics are almost there. 

Long support is currently broken, stdlib has mostly gone missing along with all the algorithm and allocation libraries - hopefully I can get the regular malloc back in. 

The old c based printf (mini version) has reappeared - writing printf in assembler once is enough for me - so bits of the test suite are now working.

zx70, 2019-07-17:
Quite obvious, but it is too funny to resist !
This one is tested:

	._ldir
		push		af
	.ldir_loop
		ld		a,(hl)
		ld		(de),a
		inc		hl
		inc		de
		dec		bc
		ld		a,b
		or		c
		jr		nz,ldir_loop
		pop		af
(ret)

It is 11 bytes long, probably it is worth to keep it as a subroutine and save space.
For a better emulation, the flags after the LDI*/LDD* instructions should be altered as follows:

	S - Sign
	Z - Zero flag 
	F5 - undocumented flag 
	H - Half Carry 
	F3 - undocumented flag 
	P/V - Parity or Overflow 
	N - Subtract 
	C - Carry 

	SZ5H3VNC
	--*0**0-

PV set if BC not 0
F5 is bit 1 of (transferred byte + A)
F3 is bit 3 of (transferred byte + A)

suborb, 2019-07-17:
@zx70 yes, I did some magic in z80asm to handle mapping:

- jr cc, label -> jp cc,label
- djnz label -> dec b; jp nz,label
- ldir/ldi/ldd/lddr/cpi/cpir/cpd/cpdr -> __z80_asm_XXXX

The jumps work well as long as they don't reference ASMPC, and the emulation is the same as we did for the Rabbit. So there's an ldir implementation here: https://github.com/z88dk/z88dk/blob/feature/8080/src/z80asm/dev/z80asm_lib/ldir.asm

suborb, 2019-07-17:
Comment to indicate the things that have been removed so I can document them later:

- adt: All
- alloc: Balloc
- strings: strlcat, strlcpy, memops, memopi, memopd
- stdlib: _Number conversion routines_, qsort, bsearch, extract_bits, inp, outp, 
- sccz80: builtin functions
- stdio: funopen() and device support, scanf, most printf format options

SDCC support

The only critical feature removed is number conversion routines. I'll see if I can resurrect some versions from the grave/adapt the newlib ones without too much pain.

suborb, 2019-07-17:
This also opens up the idea of 8085 support. Some of the undocumented opcodes look interesting:

* DSUB - 16 bit subtract HL - BC
* ARHL - Arithmetic Right shift HL
* RDEL - Rotate (shift) DE Left into carry, fill in zero
* LDHI - Load DE with HL plus Immediat8 (HL+i8->DE)
* LDSI - Load DE with SP plus Immediat8 (SP+i8->DE)
* SHLX - Store HL indeXed by DE [DE]=HL
* LHLX - Load HL indeXed by DE HL=[DE]

SHLX is basically `l_pint` for example. I'm not sure if DSUB takes into carry, but worth thinking about.

zx70, 2019-07-17:
That's a cute one.  I'd really like to try to add support for the Kyocera family handheld computers (aka Tandy 100/200)

suborb, 2019-07-17:
Yeah, I think this and any 8085 enhancements will open an entire new tranche of targets. Getting to 100 z80 targets feels like it's a long way off.

zx70, 2019-07-17:
we are totally crazy  :D

suborb, 2019-07-18:
Let’s call this one done for now.

closed 2019-07-18 by suborb

suborb, 2019-10-07:
Ken Yap (the very first version of z88dk was based off his zcc096.zip package), has a wip 8080 port of sdcc in this repo: https://github.com/kenyapcomau/sdcc-8080