z88dk/z88dk #1229: cp/m - inconsistent carriage return - new line behaviour

feilipu, 2019-07-16:
Recently, I found that the CP/M `fprintf()` behaviour is different between the two libraries. And, I think they're both probably not perfect. :sweat_smile:

```C
// for cpm target using sccz80 C compiler and classic library, run test.com
// zcc +cpm test.c -o test.com

// for rc2014 target using sdcc C compiler and new library, rename test_CODE.bin and run as test.com
// zcc +rc2014 -subtype=cpm test.c -o test

void main(void)
{
    fprintf(stdout,"Hello World R\nHello World C\nHello World 2\nHello World 0\nHello World 1\nHello World 4\n");
}
```
For the hello world case, the classic library issues `0D0Ah`, and the new library issues `0A0Dh`, when `fprintf()` sees a `\n`. The result is that [in the terminal I'm using](https://wiki.gnome.org/Apps/Moserial) the classic library "double spaces" the result, while the newlib issues (probably more correctly) single line spaces.

```C
void main(void)
{
    fprintf(stdout,"Hello World R\rHello World C\rHello World 2\rHello World 0\rHello World 1\rHello World 4\r");
}
```
If the code is changed to `\r` then the behaviour is different. The classic library issues `0Dh` which produces the correct single line spacing, and the new library issues nothing (which means that there is an incorrect line of characters).

zx70, 2019-07-16:
I bumped into a similar mess some time ago,  the different way to deal with cr/lf can be confusing. 
the libraries can be reversed,  I think. .  but I'd leave Dom alone trying to give a good energic shake to the cp/m target (and possibly the whole z88dk project)  with the support for the 8080 cpu, first.

feilipu, 2019-07-17:
The small change in #1230 is enough to get the Startrek nav computer print the SRS and LRS scan correctly, with 32-bit math in classic. But, if a big change is warranted, then bring it on :+1:

zx70, 2019-07-17:
Got it.   My troubles were mostly on the console input, sorry for having increased the entropy as usual.

feilipu, 2019-07-17:
Yes. Mostly in the previous comment refers to the input issues, I found later. Added underscores, etc.

closed 2019-07-17 by feilipu

zx70, 2019-07-17:
The added underscore is a way to display a cursor. Worked very well in some case but it is becoming annoying recently

feilipu, 2019-07-17:
Yes. I saw both of those issues with Startrek. The underscore is annoying. And finally (after a few commands) it stopped, and I guess it was because I was only sending a LF it.

Totally unrelated, but the CP/M versions I've used with RC2014 don't like getting CR/LF. The CCP balks because it thinks it is getting two commands, so cuts the first command short. A symptom is the DIR (an integrated command) only printing the first file found after a CR/LF. But, that's not a z88dk issue.