z88dk/z88dk #1231: 8085 support

suborb, 2019-07-17:
Split out from #1226 - we should be able to do something interesting with the undocumented opcodes on the 8085:

* 0x08 - 10 clocks - sub hl,bc
* 0x10 - 7 clocks, - sra hl
* 0x18 - 10 clocks - rl de
* 0x20nn - 10 clocks - ld de,hl+nn
* 0x38nn - 10 clocks - ld de,sp+nn
* 0xd9 - 10 clocks, ld (de),hl
* 0xed - 10 clocks, ld hl,(de)

The standard sccz80 (`ld hl,00nn; add hl,sp; call l_gint`) could become:

```
ld de,sp+nn
ld hl,(de)
```

zx70, 2019-08-12:
http://www.classiccmp.org/cpmarchives/cpm/Software/WalnutCD/simtel/sigm/vols100/vol140/speedup.c

This ancient hack for the BDS C compiler looks interesting.

zx70, 2019-11-23:
Reverse engineering the T100,KC85,M10
https://github.com/z88dk/techdocs/tree/master/targets/m100

suborb, 2019-11-23:
Can I take this to mean that you're quite close to adding support for these machines??

I presume it's a single precision microsoft basic?

zx70, 2019-11-23:
sadly I'm not close,  I just began looking at it
I have plenty of basic entry locations I'm putting together to compare the different models (perhaps useless) .
The binary format can be put in "external roms" or in .co files.
the latter is not very efficient because the file storage is.. the ram itself, so it is not ideal for big programs to have it twice in memory.
It seems a "floppy.co" utility could help,  but it is something I'm still trying to understand. 
To be honest I just began to share the tech detail because I thought it was a waste keeping it for myself, as usual I'm not jealous if you want to jump in!

zx70, 2019-11-24:
btw most of the data I have is still under preparation, I think it will finally be in pdf or spreadsheet format

zx70, 2019-11-28:
https://github.com/z88dk/techdocs/blob/master/targets/m100/m100.asm

A reasonably readable m100 disassembly  ;)

zx70, 2019-11-29:
File updated,  I'm very close to be able to rebuild a binary block and compare it to the original rom.

zx70, 2019-12-04:
Disassembly ready, reasonably documented (but for the HW ports).
The maths was already at an advanced stage, it is the same implementation later used for the MSX (both single and double precision).  The float constants are almost identical but a small difference in the ATN table.

jhoger, 2019-12-10:
Sorry if it's bad form, didn't know how to contact zx70 but I  (John Hogerhuis jhoger@pobox.com ) can answer questions about the Model 100 (jhoger@pobox.com) offline if it helps.

A cross platform set of ROM entry points across clones already exists you could use/adapt.
The RAM file system is only 32k (or less) but it's regularly used for .CO packaged executable.
The Model 100 also has an Option ROM area. There is currently available hardware called REX for switching between multiple ROMs and copying RAM images to/from ROM.
There is a decent disassembly available in VirtualT.
There is info on the RAM filesystem at Bitchin100.com and in the NEC Reference Manual (information is applicable to the M100.
Also worth mentioning that the "Model T" is a family of computers and clones (M100, T102, T200, Kyocera KC-85, NEC-8201A, NEC-8300, Olivetti M10)

There is a community around these machines that meets on the Bitchin100 mailing list, several programmers. You don't need to reverse engineer unless you want to :-)

http://lists.bitchin100.com/listinfo.cgi/m100-bitchin100.com

Excited to see z88dk target the m100 , thank you.

zx70, 2019-12-10:
Thanks for pointing it out,  but I'm using those documents as a reference already.
What I'm willing to get is a consistent cross reference between the models, including the olivetti m10 etc.
They are close relatives, but the nec model is based on a different BASIC version.    All are surprisingly close to the MSX BASIC, much more than any other msbasic version I've ever met.   The nc100 has extra features (what is the 'wand'?) Which weren't available elsewhere,  so the memory locations change.

ieure, 2019-12-10:
> The nc100 has extra features (what is the 'wand'?)

Probably related to [the M100 barcode reader](https://munk.org/typecast/2014/08/02/trs-80-bar-code-reader-drivers-for-model-100200-and-kyotronic-85/).

jhoger, 2019-12-10:
I think NC100 is Amstrad and not related to the Model 100. Maybe you mean the 8201A. The ROM for the NEC 8201A is very similar to the Model 100. It is a variant of the Microsoft BASIC ROM in for the TRS-80 Model 100. Willard Goosey maintains assembler header files for the different models.

The main difference between NEC8201A and M100 is the NEC has support for 64K of RAM in two 32K banks, or I think it can do an all-RAM mode. The NEC8300 comes with the full complement of 32K. Also, it has XMODEM for transferring text files only. The M100 has a LINE command, but the NEC does not.

As to barcode wand, the M100 has a dedicated barcode port too. It requires a driver to be loaded, I assume the NEC does as well.

"The binary format can be put in "external roms" or in .co files.
the latter is not very efficient because the file storage is.. the ram itself, so it is not ideal for big programs to have it twice in memory."

Yes. But you can load a file straight from disk to the address it is assembled to run at. Usually people use a "Tandy Portable Disk Drive" client like TS-DOS running in OptROM to load large files directly.

You don't actually need a TPDD there are TPDD services. I am the author of LaddieAlpha disk emulation service,  and there are others like mComm.

You can retain a CO file in high RAM if you keep the space CLEAR'd for it. Then what people do is create a small BASIC trigger file to EXEC it. This common when using TEENY, a 800-byte TPDD client. Then you don't have the double usage issue.

But definitely if you're deciding what target formats to build, .CO and OptROM images are the best.

There was some discussion on the list as to creating a special tokenized .BA file as a container for machine language code. The ML would be hidden in the high numbered unlistable BASIC lines. Either the ML code would have to be position independent (there are (expensive) ways on the 8085) or it would have to patch up the jumps to locate the code before running.

zx70, 2019-12-11:
Sorry!  Too many codes,  I meant m100

zx70, 2019-12-11:
>or it would have to patch up the jumps to locate the code before running.

Here z80asm could help. I don't know if it still available in the recent versions,  but it could build self relocating code.
Btw thanks for suggesting the preferred formats.

zx70, 2020-02-04:
I chose the less optimal approach, but had a bit of fun  :)    Now the asm source can generate binary exact roms for m100, m10 and kc85.  The resulting map is a great reference :)

suborb, 2020-02-10:
@zx70 I've set this branch up for you: https://github.com/z88dk/z88dk/tree/feature/m100

- The crt0 is totally wrong (it's basically a ROM at the moment)
- No implementation so fill in the bits in target/m100/stdio/ etc

zx70, 2020-02-17:
I'm slowly proceeding :)
The prototype  (tandy m100 only) is capable of minimal console support, wrong new line, no cls.  The generated .co file needs to be loaded by BASIC, with an initial CLEAR statement (e.g. clear 0, 39999).
Then runm "a.co" should work

zx70, 2020-02-27:
new default position for CLEAR: 49999
minimal documentation is now in the wiki

suborb, 2020-03-01:
I've updated sccz80 and the crt0 library to use some 8085 undocumented codes. The m100 target has been updated to use them so it should be marginally faster than when just using 8080 opcodes.

I'll take a look at the main library and see if there's much point adding an 8085 variant - I'm suspecting that we may only gain a small amount for the C library routines, but given the small number of them it may not be worth the extra build time.

closed 2020-07-06 by suborb