z88dk/z88dk #1239: testsuite failing in branch z80asm_macros

pauloscustodio, 2019-07-19:
@suborb, I need your help.

In the branch z80asm_macros the testsuite is failing after I have:
1) created a new '#' string-ize operator in z80asm to convert the following operand to a string;
2) removed the old '#' constant expression operator;
3) replaced all instances of '#(' with '+(' and '#-(' with '-(' in the library, rules and testsuite.

The testsuite fails with different code generated, that apparently is not related to the changes I have done, e.g. in mathops_ieee.opt:
Output is
```
	xor	a
	ld	(hl),a
	inc	hl
	ld	(hl),a
```
and results expected is
```
	ld	(hl),+(0 % 256)
	inc	hl
	ld	(hl),+(0 / 256)
```
The output is shorter than the result expected.
Is this some side-effect? I did not find a corresponding rule in the rules files.

Thanks in advance.

suborb, 2019-07-19:
That optimiser can be quite good at times!

I'll take a look this weekend, I think I ran into similar problems last time I tried to remove the '#'

suborb, 2019-07-20:
There's a rule in z80rules.1 line 3815 that's firing and doing this replacement.

pauloscustodio, 2019-07-22:
Next testsuite issues, also in mathops_ieee.opt:

1) Output and expected do not have the same stack effect. Output is:
```
	ld	hl,4	;const
	add	hl,sp
	call	l_glong
	pop	bc
	pop	bc
	pop	bc
	pop	bc
	ret
```
expected is:
```
	pop	bc
	pop	bc
	ld	hl,0	;const
	add	hl,sp
	call	l_glong
	pop	bc
	pop	bc
	ret
```

2) Output has an extra call to l_f32_sint2f; output is:
```
	ld	hl,0	;const
	ld	de,16256
	call	l_f32_sint2f
	ex	de,hl
	ex	(sp),hl
	push	de
	call	l_jphl
	pop	bc
	pop	bc
	pop	bc
	pop	bc
	pop	bc
	ret
```
Expected is:
```
	ld	hl,0	;const
	ld	de,16256
	ex	de,hl
	ex	(sp),hl
	push	de
	call	l_jphl
	pop	bc
	pop	bc
	pop	bc
	pop	bc
	pop	bc
	ret
```

suborb, 2019-07-22:
I’m not seeing those. They look like codegen rather than optimisation rules - have you rebuilt sccz80?

pauloscustodio, 2019-07-22:
I will try to rebuild. I've now split the removal of the # operator to a separate branch.

pauloscustodio, 2019-07-24:
sccz80 is outputing wrong code in my mingw environment, but correct in Ubuntu. I'll try to track it down.

closed 2020-08-14 by suborb