z88dk/z88dk #1241: (g800) Text output of built-in font is incorrect

fu-sen, 2019-07-19:
It has a different possibility from the conventional report with g800, and I was a new issue.
But that is a simple and obvious bug.

chr.c:
```
#include <stdio.h>

void main() {
  printf(12);

  for (int c=32; c<127; c++) {
    printf("%c",c);
  }

  while(1){}
}
```
```
zcc +g800 -create-app -bn chr.ihx chr.c
```
and
```
zcc +g800 -create-app -clib=g850b -bn chr.ihx chr.c
```
[image]

If "-CLIB=g850", it can be displayed correctly:
```
zcc +g800 -create-app -clib=g850 -bn chr.ihx chr.c
```
[image]

fu-sen, 2019-07-20:
I was worried about the output character deviation, so I tried with a real Pocket Computer PC-G850V:
[image]
It produces slightly different output than the g800, but has the same problem.

fu-sen, 2019-07-21:
I was able to reproduce the same display using IOCS call.
When I look closely at the output results, "RUN" is lit. That was a hint.
This part lights up when X coordinate is 24 or more

chrtest.c:
```
void put_chr(char posx, char posy, char chr)
{
#asm
    ld   ix,0
    add  ix,sp
    ld   a,(ix+2)       ; chr
    ld   d,(ix+4)       ; posy
    ld   e,(ix+6)       ; posx
    call 0be62h
#endasm
}

int key_codewait()
{
#asm
    call 0bcfdh
    ld   l,a
    ld   h,0
#endasm
}

main()
{
  for (int c=32; c<127; c++)
  {
    int x=(c-32)%64;
    int y=(c-32)/64;
    put_chr(x,y,c);
  }

  key_codewait();
}
```
```
zcc +g800 -create-app -o chrtest.ihx chrtest.c
```
[image]

Therefore, in the current Nightly build, the width of the text is incorrect at least other than "-CLIB=g850".

zx70, 2019-07-22:
I think I found what happened.   To have a simple, generic support I was avoiding put_chr.
I simply replaced it with put_string, but outputting only one character.
it worked but the VT-ANSI emulation was not full (yet having the whole code inside to deal with coordinates).
I changed it, not remembering about this mess with the many different text resolutions (E200, G800, G815).
I'm now reverting it to the generic implementation, so different text resolutions, scrolling, etc won't be a problem anymore and some text program will be fairy usable on all the models without needing to recompile it.

zx70, 2019-07-22:
I finally chose not to revert, the smaller resolution will still work also on the G850, but the text scrolling will happen sooner.
Please test again an report the results.

fu-sen, 2019-07-22:
Yeah. It can be hoped for a solution.
I will test at Nightly Build after tomorrow.

fu-sen, 2019-07-23:
My PC was updated to nioghtly Build 20190723 and I tested. It works well.

(24x4)
[image]

-CLIB=g850 (36x8)
[image]

-CLIB=g850b (24x6)
[image]
change "for (int c=32; c<192; c++)"
[image]

That is the perfect solution. This issue can be closed. Thank you!

closed 2019-07-23 by fu-sen