z88dk/z88dk #1254: [question] How to compile and link programs for custom z80 machine

kodiak77, 2019-07-27:
I have some questions about bare metal programming using z88dk.

I have a custom z80 system and I am trying to write a program/os using three sections of code: one for system startup (address 0x0), one for handling interrupts (address 0x38) and one for the rest of the code (should follow interrupt section).

I want to avoid all libraries and CRTs for now. I managed to program such bare system using SDCC toolchain and managed to get assembly and C interoperability. Now I am trying to do it with z88dk.

I could not figure out completely how to do it using z88dk. I got stuck figuring out how to link different sections for particular addresses. So far I found [this link](https://github.com/z88dk/z88dk/issues/247) which informed me that a custom memory map is needed to place code sections appropriately.

Based on this I created `memory_map.asm`:
```assembly
SECTION crt0
ORG     0x0 

SECTION isr 
ORG     0x38
```

I beleive all other sections created by compiler and assembler should automatically follow these two and don't have to be mentioned specifically.

Sections `crt0` and `isr` are defined in following assembly files:
`crt0.asm`:
```assembly
    SECTION crt0

    extern  _mainc

    ld      sp, 0xFFFF

    call    _mainc

    halt
```

`isr.asm`:
```assembly
    SECTION isr

    extern  _isrc

    call    _isrc

    reti
```

`mainc` and `isrc` are written in C and should be put in the section behind the interrupt section. They are defined in the file `code.c`:
```c
void mainc(void) {}

void isrc(void) {}
```

Finally, I wrote the following Makefile:
```make
.PHONY: all clean

SOURCES = $(wildcard *.c) $(wildcard *.asm)
OBJECTS = $(addsuffix .o,$(basename $(filter-out memory_map.asm,$(SOURCES))))

all: $(OBJECTS)
    z80asm -s -l -m -g --cpu=z80 -d -v -b memory_map.asm $(OBJECTS)

%.o: %.c
    zcc +embedded -compiler=sccz80 -v -m -g -s -nostdlib --no-crt -clib=new -c $<

clean:
    @rm -f *.o
    @rm -f *.lis
    @rm -f *.bin
    @rm -f *.sym
    @rm -f *.map
    @rm -f *.reloc
    @rm -f *.def
    @rm -f *.err
    @rm -f *.ihx
```

When make is called, the verbose output says that memory_map.bin, memory_map_crt0.bin and memory_map_isr.bin are created. The former is not created, only crt0 and isr binary files exist. I expect that at least one more binary should be created for sections that follow interrupt section.

Also, if I try to write another file with code, e. g. `another_code.c`:
```c
unsigned int e;

void f(void)
{
    unsigned int a = 1, b = 2, c = 3, d;
    d = a + b + c;
}
```

I get an error:
```
Error at file 'another_code.c' line 29: symbol 'l_gintspsp' not defined
Error at file 'another_code.c' line 32: symbol 'l_gint' not defined
```

I suspect I am missing some basic CRT functionality. I probably have to study CRTs for existing platforms and figure what is minimum necessary functionality for C to work.

I would really appreciate some help or pointers to documentation that I missed. Please ask for any clarifications if I didn't explain anything enough.

feilipu, 2019-07-28:
The issue causing the error is that the `sccz80` or indeed the `zsdcc` is expecting to have a library of stuff to use when compiling the C code. Both `sccz80` and `zsdcc` use slightly different linkage language for their libraries, so we have to build both types of linkages when building a library in z88dk.

The library of stuff is to do things like; addition `+`. The compiler doesn't do addition by itself. It calls a standard library function to do the addition, and you've asked that the standard library not be made available.

I suggest that the error can be removed by removing ` -nostdlib` from your command line. The ` --no-crt` option is enough to stop initialisation code.

In your memory map, you should have some standard `SECTION` names for code, data, and bss. These can be found by looking at the [memory maps for z80](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/crt_memory_model_z80.inc) for example. Using the `org` of `-1`  will ensure that these items are concatenated.

You are using the `+embedded` target which is great. There is [a lot of documentation](https://www.z88dk.org/wiki/doku.php?id=temp:front#c_library_reference_new) about this target and the new library, and therefore (if you don't turn everything off) then you can get a lot of info by adding `-v`, `--list`, and `--c-code-in-asm` options to your commands, and reading the resulting verbose files.

Rather than using a make file to build all the pieces, try to use a `.lst` file which has your memory map first in the file, and then all the other assembly and C files. Here's [an example](https://github.com/RC2014Z80/RC2014/blob/master/ROMs/CPM-IDE/sio/cpm22.lst) I prepared earlier. Leave the suffixes off the file names, as `zcc` will work this out automatically.

I think (I'd ask others to confirm) that you're actually making life more complex by using `make`. It prevents `z88dk` from being aware that all the calls to `zcc` that your make file is producing are actually for _one_ build, and they all need to be linked together. As a matter for comparison, I've not needed to use `make` for anything I've done with z88dk to date, not to say that `make` isn't needed at all.

Producing separate files for the `crt0` and `isr` is a bit wasteful, since they are both usually located in the first page of memory. To organise code there is the `ALIGN` word which will place code at specific absolute or relative locations within a `SECTION`. For example `ALIGN 0x100` or `ALIGN 256` will place code at a page boundary. `ALIGN 0x0038` will place code at the INT location. `ALIGN 0x0066` will place it at the NMI location.

There is also the `PHASE` word which will instruct the compiler to assemble following code as if it were located elsewhere. This is useful where you have some [code that will be relocated](https://github.com/RC2014Z80/RC2014/blob/master/ROMs/CPM-IDE/sio/cpm22bios.asm#L79) from ROM to RAM on initialisation. It is terminated by `DEPHASE`, unsurprisingly.

Hope that's useful.

aralbrec, 2019-07-28:
The newlib has a target specifically for bare z80 systems which feilipu is talking about above but he missed mentioning the main documentation which is here:

aralbrec, 2019-07-28:
About separate compilation of files in a makefile:

As felipu mentioned this is not really necessary as zcc is able to generate an output binary from a list of files either specified in the compile line itself or in a separate file listing all files in the project.  zcc knows the difference between .c, .asm, .o, .c.m4, .asm.m4, etc and will properly generate object files for all file types before linking them together to generate the output.

This compile line does both things:

zcc +z80 -vn -O3 -clib=new test.c asmroutine.asm objfile.o @manyfiles.lst -o test -create-app

"test.c", "asmroutine.asm", "objfile.o" are listed explicitly then more files are taken from the file "manyfiles.lst"

"manyfiles.lst"
```
test2.c
myasm.asm
res/graphics.asm.m4
```

Of course such a compile line rebuilds the output from scratch on every invoke whereas a makefile would only rebuild files that have changed.  This can make a difference when using zsdcc as compiler as that compiler can be very slow for large c files and high optimization enabled.

If you're using a makefile, the intermediate step is generating an object file so "-c" should appear on the zcc line as you have done with the slightly wrong invoke syntax in your makefile (see the first link for examples of correct invoke).  The linking should also be done by zcc giving it the object files and not z80asm because it is at this point that the crt is gathered (yours or the library's) which is where the memory map is found.  It also keeps open the path to using appmake to automatically generate output binaries.

kodiak77, 2019-07-28:
Thank you guys a lot for comprehensive answers. I will study them and report back.

feilipu, 2020-01-22:
Ok?

closed 2020-01-22 by feilipu

kodiak77, 2020-03-07:
Sorry guys, I only recently managed to continue working on my project. Using your feedback I managed to solve the problem. Thank you again!