z88dk/z88dk #1258: CP/M - additional tools and apps

feilipu, 2019-08-06:
> @zx70 I have few sources of similar tools [to ddir] which I'm arranging just for fun, if you think they can be of any use in the CP/M world, I can send the results (sources and compiled code).

I know of quite a few people who are interested in source code that works in CP/M, particularly if it can be compiled with a modern development environment. Specifically at:
 - [rc2014-z80](https://groups.google.com/forum/?#!forum/rc2014-z80)
 - [retro-comp](https://groups.google.com/forum/?#!forum/retro-comp)

I'm sure there are many others.

Would it be sensible to add them to the [`examples/cpm`](https://github.com/z88dk/z88dk/tree/master/examples/cpm) directory?

Or better to start a new repository `z88dk/apps` and have a `cpm/tools` directory there?
Along with `cpm/games`, `cpm/editors`, etc...

zx70, 2019-08-06:
On the z88dk perspective I'd rather, say, extend the functions used by ddir to the TRS80, etc.

I noticed that some existing archive have interesting pieces of code, I remember FUZIX, the old VAX/PDP11 stuff, and several CP/M era BBS archives, TRS80 and MSX C programs to be built natively...  but I'm a bit scared to "maintain" a local repository.

@suborb ..perhaps we can add yet another repository for "ported stuff", trivia, etc.. ?
I'd surely keep them in source code form: ddir can be kept as small as possible, fully featured, adjusted for particular display needs, etc..

zx70, 2019-08-06:
https://github.com/z88dk/z88dk-ext

A parallel repository should be a comfortable place in several cases.  
I'd try ecouraging people to post (via pull requests) their creations or finds.  
I think the current repository is self explanatory, let me know what you think about it.

feilipu, 2019-08-06:
Some thoughts (mostly completely obvious).

- There's no point to collect binary CP/M applications and tools. [Humongus CP/M](http://www.classiccmp.org/cpmarchives/) and [Retro Archive](http://www.retroarchive.org/cpm/) (and others) have that well covered. And there's no point to get into licencing issues this kind of collection might cause.
- Applications should be compilable or assembleable by `zcc` and use z88dk libraries where possible and relevant.
- Source should have example `zcc` incantations in comments in the source, or in a separate readme, to make the chance of success close to 100%.
- There needs to be a licence file or licence header associated with each application.
- The Spectrum Next community will grow go from 100 machines to over 3,000 machines in the next months, so some focus on applications for the `zxn` platform would get attention.
- Probably the other largest active user base is the RC2014, with over 2,000 machines sold and they have various CP/M versions available. Though many of these RC2014 machines are MSBASIC only they are already supported by z88dk new library.

ped7g, 2019-08-07:
SW available as binary is pretty much useless, in 99% worth rewrite... (if you have time, of course, which one in 99% of cases does not have, sadly).

zx70, 2019-08-07:
So now we have it.

I'd avoid to create target specific sections, but I'm not totally against it.
I'm also unsure about data sections (e.g. vector cliparts, picture and sound libraries, etc).. let's avoid it for the moment.

closed 2019-08-07 by zx70

zx70, 2019-08-08:
I just completed sorting out my mods to SQ/USQ tools, I think the results are interesting.
The updated sources can be built with z88dk or with a modern compiler.
I also converted USQ119.ASM to z80asm, it is not yet optimized but works.  This should help the portability to other targets.

This work was originally intended for the v6z80p (aka Old School Computer Architecture).
Compared to the original version it features optional support for wildcards and output redirection or smaller size (and redirection limited to files) if built without extras.

Multiple wildcard expressions can be provided in a single shot, e.g.

C> USQ *.CQ *.DQT

zx70, 2019-08-08:
For the sake of completeness, I recovered the much simpler ASM version of the USQ program, which was 1920 bytes long.  After converting it to Z80 mnemonics with "toz80.awk" and optimizing it a little I reduced it to 1689 bytes !     Having a working source code means also that it could be ported to different targets.
I wonder what was the smallest TPA space aroud a the time, but I think that the new variants would extend the range of the compatible systems: sq.com can now be smaller 14800 bytes, while the original version was 15744 ..I wonder if one or both of them can be run on a 16K system  :)

zx70, 2019-08-11:
interesting places to pick command sources to try to port to z88dk:

http://www.ibiblio.org/pub/academic/computer-science/history/pdp-11/language/decus-C/6,1/grep.c

zx70, 2019-09-13:
Recent updates:
uuencode/uudecode
lzencode/lzdecode (formerly lzhuf.c)
grep cleaned and improved (wildcards, file recursion, more build options)

zx70, 2019-10-31:
New interesting tool,  lar.c
It does more or less what delbr11 did, but in portable C.  It can be built with z88dk or gcc and limited in functionality to reduce the program size.   Automatic usq and uncrunch plus minimal archive editing capabilities are provided.   Original skeleton found in the DEC archives and merged with compression programs portions.