z88dk/z88dk #1264: Installation of z88dk on Ubuntu 19.04 Fails with 'compress.c:22:10: fatal error: zlib.h: No such file or directory'

spectrumcomputing, 2019-08-11:
**OS: Ubuntu 19.04 Desktop**

**Issue**
I get the following errors when building the nightly build following the instructions

```
compress.c:22:10: fatal error: zlib.h: No such file or directory
 #include <zlib.h>
          ^~~~~~~~
compilation terminated.
make[6]: *** [Makefile:1705: compress.lo] Error 1
make[6]: Leaving directory '/tmp/sdcc/support/sdbinutils/bfd'
make[5]: *** [Makefile:1748: all-recursive] Error 1
make[5]: Leaving directory '/tmp/sdcc/support/sdbinutils/bfd'
make[4]: *** [Makefile:1213: all] Error 2
make[4]: Leaving directory '/tmp/sdcc/support/sdbinutils/bfd'
make[3]: *** [Makefile:2692: all-bfd] Error 2
make[3]: Leaving directory '/tmp/sdcc/support/sdbinutils'
make[2]: *** [Makefile:850: all] Error 2
make[2]: Leaving directory '/tmp/sdcc/support/sdbinutils'
make[1]: *** [Makefile:162: sdcc-sdbinutils] Error 2
make[1]: Leaving directory '/tmp/sdcc'
make: *** [Makefile:74: bin/zsdcc] Error 2
peter@ubuntu:~/z88dk$ 
```
**Steps to reproduce**

sudo apt-get install dos2unix libboost-all-dev texinfo texi2html libxml2-dev subversion bison flex
wget http://nightly.z88dk.org/z88dk-latest.tgz
tar -xzf z88dk-latest.tgz
cd z88dk
chmod 777 build.sh
./build.sh

suborb, 2019-08-11:
I’ve just updated the wiki page to include zlib1g-dev which I’m guessing is the missing package.

spectrumcomputing, 2019-08-11:
Thanks, will give that a go.

spectrumcomputing, 2019-08-11:
Thanks @suborb 

That worked.

closed 2019-08-11 by spectrumcomputing