z88dk/z88dk #1268: sccz80 - inline __asm syntax

feilipu, 2019-08-20:
Following up.

I'm seeing following errors from this code.

```c
void main(void)
{
    {
        printf("hello i2c world\n");
        call_error(12h);
    }
}
```

```sh
$ zcc +yaz180 -subtype=app -clib=new -v -m -SO3 --list -l --c-code-in-asm @i2ctest.lst -o i2ctest -create-app

PROCESSING ./i2ctest.c
z88dk-ucpp -iquote"." -D__Z88DK -D__YAZ180 -D__Z180 -D__SCCZ80 -isystem"/home/phillip/Z80/z88dk/lib/config/../..//include/_DEVELOPMENT/sccz80"  -DSCCZ80 -DSMALL_C -D__SCCZ80  "./i2ctest.c" "/tmp/tmpXXDmGJp4.i2"
z88dk-zpragma -sccz80 < "/tmp/tmpXXDmGJp4.i2" > "/tmp/tmpXXDmGJp4.i"
sccz80  -ext=opt -mz180 -standard-escape-chars  -cc  "/tmp/tmpXXDmGJp4.i" -o "/tmp/tmpXXDmGJp4.opt"
sccz80:"./i2ctest.c" L:31 Error:Unknown symbol: __asm
sccz80:"./i2ctest.c" L:31 Error:Expected ';'
sccz80:"./i2ctest.c" L:31 Error:Unknown symbol: rst
sccz80:"./i2ctest.c" L:31 Error:Expected ';'
sccz80:"./i2ctest.c" L:31 Error:Expected ';'
sccz80:"./i2ctest.c" L:31 Error:Unknown symbol: h
sccz80:"./i2ctest.c" L:31 Warning:Implicit definition of function 'call_error' it will return an int. Prototype it explicitly if this is not what you want. [-Wimplicit-function-definition]
sccz80:"./i2ctest.c" L:31 Error:Missing token, expecting ) got h
sccz80:"./i2ctest.c" L:31 Error:Expected ';'
sccz80:"./i2ctest.c" L:31 Error:Expected ';'
sccz80:"./i2ctest.c" L:31 Error:Invalid expression

Maximum (10) number of errors reached, aborting!
Compilation aborted
$ 
```
when [sccz80 specific inline asm format is not provided](https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/proto/arch/yaz180.h#L122). When it is, things work perfectly.

Following up on [comment here](https://github.com/z88dk/z88dk/commit/ea82cd55098a10d70f5f8764d9a2ecbd1ff25b4d#r34747696).

zx70, 2019-08-22:
are you putting the semicolon after __endasm; ?

zx70, 2019-08-22:
also try keeping __asm and __endasm at the leftmost column

suborb, 2019-08-22:
I popped in to ask the original question and now I’m away again. However I think the issue is that the asm markers aren’t left aligned as Stefano said above. 

Given that sdcc supports them anywhere, zpragma should too. I’ll fix it up when I get back.

suborb, 2019-08-28:
Some things are never simple, the output of `-E` for `-compiler=sccz80` is:

```
void main(void)
{
    {
        printf("hello i2c world\n");
         do{ __asm rst 8h ; call_error( 12h ) defb 12h  __endasm; }while(0) ;
    }
}
```

And for `-compiler=sdcc` is:

```
void main(void)
{
    {
        printf("hello i2c world\n");
        do{ __asm <87> rst 8h ; call_error(12h) <87> defb 12h <87> __endasm; }while(0);
    }
}
```

Some embedded \x87 characters separating lines in the macro expansion.

zx70, 2019-08-28:
ouch

feilipu, 2019-08-28:
What is even` \x87` ?
It looks like `BEL` with the high bit set..?

suborb, 2019-08-30:
I think it’s just a randomly chosen character that was used to fix an issue preventing multi-line assembler macros. The compiler maps it into a newline when reading the preprocessed output. 

I’m not too keen on emulating this one since we have alternate ways of doing it.

closed 2019-09-14 by suborb