z88dk/z88dk #1270: sccz80 - Weird padding in function pointer

teiram, 2019-08-31:
Using z88dk 1.99C
I have a type defined as:

```
typedef struct _screen_handler { 
void (*enter)();
void (*exit)();
uint8_t (*update)();
} screen_handler_t;
```

And an array of such elements declared as:

```
const screen_handler_t screen_handlers[] = {
 {games_enter, games_exit, games_update},
 {categories_enter, categories_exit, categories_update},
 {screenshots_enter, screenshots_exit, screenshots_update}
};
```

I have debugged the resulting app, compiled with options:

`CFLAGS = +msx -m  -vn -O3 -create-app -subtype=rom`

and have detected that each of the elements in the array is padded with a zero byte at the end, so that the array has a total size of 21 bytes. Of course trying to access the elements of the array by index other than zero, crashes the application because of the misleading offsets.

I was using an older version and didn't get this issue.

suborb, 2019-08-31:
I've gone back to z88dk 1.99C and I can't reproduce this one with the following test program when running on MacOS, which platform are you using?

```
extern void games_enter();
extern void games_exit();
extern void games_update();

extern void categories_enter();
extern void categories_exit();
extern void categories_update();

extern void screenshots_enter();
extern void screenshots_exit();
extern void screenshots_update();

typedef struct _screen_handler {
void (*enter)();
void (*exit)();
void (*update)();
} screen_handler_t;

const screen_handler_t screen_handlers[] = {
 {games_enter, games_exit, games_update},
 {categories_enter, categories_exit, categories_update},
 {screenshots_enter, screenshots_exit, screenshots_update}
};
```

If the above test program above generates the following:

```
        SECTION rodata_compiler
._screen_handlers
        defw    _games_enter + 0
        defw    _games_exit + 0
        defw    _games_update + 0
        defw    _categories_enter + 0
        defw    _categories_exit + 0
        defw    _categories_update + 0
        defw    _screenshots_enter + 0
        defw    _screenshots_exit + 0
        defw    _screenshots_update + 0
        SECTION code_compiler
```

Then something else is going on and we'll need to investigate further.

teiram, 2019-09-01:
I'm running on a Debian GNU/Linux 9 64 bits , hosted in a Windows 10 WSL.
Could you provide simple instructions to do the same test on my side and see and I have a different result. I actually found out the problem by debugging on a machine simulator (openmsx) but it would be interesting to see what result do I get from your same test.

suborb, 2019-09-01:
Sure, sorry, `zcc +msx -a test.c` will generate a .c.asm file which should have code looking like the above in it.

teiram, 2019-09-01:
I have detected that the problem appears when the function is defined to return something. This:

```
typedef struct _screen_handler {
void (*enter)();
void (*exit)();
uint8_t (*update)(); 
} screen_handler_t;

```
Generates:

```
        SECTION rodata_compiler

._screen_handlers
        defw    _games_enter
        defw    _games_exit
        defw    _games_update
        defb    0
        defw    _categories_enter
        defw    _categories_exit
        defw    _categories_update
        defb    0
        defw    _screenshots_enter
        defw    _screenshots_exit
        defw    _screenshots_update
        defb    0
```

suborb, 2019-09-01:
Your output is quite a bit different to mine - it's missing the +0 at the end of the defw lines.

This means it precedes f3012bbe89051dde88783d0b7cd46c5fc2a3ce3f (issue #569) which was in January 2018 - can you check what version of sccz80 is running (it's printed at the top of the .asm file).

teiram, 2019-09-01:
It says: 

 Version: 9193-a9bca3d-20170717  

Looks like it is too old, isn't it? But I have version 1.99C installed. Could the compiler have being found in a different place?

suborb, 2019-09-01:
It’s a little old! I wonder if you did a make install on an older version which is being found first in the path?

I don’t know how you installed but the usual way is just to your home directory and then modifying the path in .profile. Depending on where in the file you’ve made the modification it may be that /usr/local/bin has been prepended. 

Running sccz80 with no arguments should print its version.

teiram, 2019-09-01:
It is quite strange. I see that:
[image]

teiram, 2019-09-01:
Probably I messed something up. The changelog in my source dir says:
[image]
Looks like this is some nightly build or something similar? I will download the 1.99C release from the GitHub page and see how it turns out.

Thanks for all your support!

teiram, 2019-09-04:
I just confirmed that my z88dk was outdated and was not 1.99C, probably a previous nightly build.
I can see that the weird zero paddings are not there anymore. 
Thanks a lot!

closed 2019-09-04 by teiram