z88dk/z88dk #1271: Issue with

teiram, 2019-09-04:
I don't know exactly what is going on, but some code that was working before doesn't work anymore. 
Basically I think that I reduced the problem to some issue with parameter passing, this is the relevant code:

```
typedef struct {
        uint8_t y;
        uint8_t x;
        uint8_t pattern;
        uint8_t color;
} sprite_attribute_t;

static sprite_attribute_t sprite;

sprite_attribute_t *my_put_sprite(uint8_t id, uint8_t x, uint8_t y, uint8_t pattern,
        uint8_t color, sprite_attribute_t *sprite) {
        sprite->x = x;
        sprite->y = y;
        sprite->pattern = pattern;
        sprite->color = color;
        msx_vwrite(sprite, 6912 + (id << 2), 4);
        return sprite;
}
```

In the generated code I see something like:

```
; Function my_put_sprite flags 0x00000200 __smallc
; struct 0__anonstruct_15 sprite_attribute_t*my_put_sprite(unsigned char id, unsigned char x, unsigned char y, unsigned char pattern, unsigned char color, struct 0__anonstruct_15 sprite_attribute_t*sprite)
; parameter 'struct 0__anonstruct_15 sprite_attribute_t*sprite' at 2 size(2)
; parameter 'unsigned char color' at 4 size(1)
; parameter 'unsigned char pattern' at 6 size(1)
; parameter 'unsigned char y' at 8 size(1)
; parameter 'unsigned char x' at 10 size(1)
; parameter 'unsigned char id' at 12 size(1)
._my_put_sprite
        pop     bc
        pop     hl
        push    hl
        push    bc
        inc     hl
        ex      de,hl
        ld      hl,12   ;const
        add     hl,sp
        ld      a,(hl)
        ld      (de),a
```
I've run with a debugger and it seems that DE is pointing 1 byte passed the address of sprite. In my binary sprite is at position $C074 and I'm seeing something like the following:

[image]

$407C is the address of my_put_sprite.
$C074 is the address of sprite.
As you can see after the first write (LD (DE), A) , I get a "1" into $C075.

Since I'm calling the function as:
`my_put_sprite(1, 2, 3, 4, 5, &sprite);`

I understand that I should get a "2" into $C074, a "3" into $C075,... so it seems that something gets shifted somehow.
I'm doing something wrong?
Out of curiosity, is there any reason why the elements in the stack always seem to use 2 bytes?

Thanks
Manuel

suborb, 2019-09-04:
This has already been fixed as #1126 

We're due a release soon - there's just a final bit of work to do with math32 in newlib holding it up - so if you could switch to a nightly this issue will be fixed.

With regards to the argument pushing - that's always been the __smallc convention so it's stuck and there's a lot of code around that depends on it.

There's quite a few calling conventions that are detailed here: https://github.com/z88dk/z88dk/wiki/CallingConventions including one that matches the SDCC char passing one.

teiram, 2019-09-04:
Thanks for the info. I will try with a nightly build. Would it be enough to clone master and build or does development take place in another branch?

suborb, 2019-09-04:
Just clone master.

teiram, 2019-09-04:
Just did it but build fails. I don't know if I'm lacking any new dependency (1.99C compiles cleanly with this same setup)

[image]

teiram, 2019-09-04:
OK. Found it. I installed uthash-dev package and compilation progresses again

teiram, 2019-09-04:
This one seems trickier:
[image]

Any idea?

suborb, 2019-09-04:
I think you're missing the submodules (certainly that covers uthash) - add them with:

```
git submodule update --init --recursive
```

Hopefully that sorts out the lvalue error as well.

teiram, 2019-09-05:
After adding submodules and regenerating my program, it finally works again as expected

Thanks a lot for the support 
Regards

Manuel

closed 2019-09-05 by teiram