z88dk/z88dk #1272: Problem in msx_vwrite_direct

teiram, 2019-09-04:
I think I've found a bug in the msx_vwrite_direct MSX asm function. On using it to handle MSX sprites, I got severe screen corruptions. The function reads like:

```
msx_vwrite_direct:
_msx_vwrite_direct:
        push    ix
        ld      ix,4
        add     ix,sp

        ld c, (ix+0)    ; count
        ld b, (ix+1)

        ld l, (ix+2)    ; dest
        ld h, (ix+3)

        ld e, (ix+4)    ; source
        ld d, (ix+5)

        ld      ix,SETWRT
        call    msxbios

       ld l, (ix+0)    ; count
       ld h, (ix+1)
...
```
The last loading of the count of bytes on HL cannot work, because IX is previously used to set the BIOS call. Therefore, the count of written bytes is not the expected one.

I've just changed that two final LD commands with:

 ```
       ld l,   c         ;count
        ld h,   b
```

and my program works again. I just restored the value from BC because the BIOS call doesn't modify BC.

Could you please provide this fix or another equivalent one?
Thanks and best regards

Manuel

suborb, 2019-09-04:
Thanks for reporting - I’ve added in your fix.

teiram, 2019-09-04:
Great. Thanks!

closed 2019-09-04 by teiram