z88dk/z88dk #1274: (classic) Graphics should work in any mode

suborb, 2019-09-10:
To the end-user it should be transparent - so we should by default link in drivers for all available screen modes - these can be pruned out manually using pragmas.

The following have already been done:

- [x] PC6001
- [x] SPC-1000 including TMS9928 modes
- [x] Other MC6847 machines
- [x] Galaksija

The following definitely haven't been done:

- [ ] CPC
- [ ] TS2068

suborb, 2019-11-05:
It looks like I changed the semantics of pixeladdress for several ports with the result that the generic putsprite/stencil no longer work: I had the screen address being returned in hl rather than de.

zx70, 2024-05-31:
I'm possibly colliding a little here,  I just sorted out the TS2068 library to gain speed.
I noticed only in a second moment that "@target/ts2068/gfx2068hr.lst" was not in the loop anymore and that it was replaced with "@target/zx-common/timex_zxn_graphics.lst".

I'm re-including gfx2068hr, putsprite() and clga() are noticeably faster.
My new rework on clga() evolved it to clga_callee() making it a valid customization of the "wide graphics" library set, no need of the retrofit trick.
Sorry for any inconvenience, I think it could be an interesting improvement for the zxn target, too.

zx70, 2024-06-03:
I think it is now fixed. 
New MODE6 specific code for clga(), xorclga() and fillb(), fat pixels are supported.
Currently some game runs faster in MODE6 with fat pixels than in MODE0 !   Adding code optimized for MODE0 requires a lot of redundant code,I think it is better to leave it on the classic ZX Spectrum library only.

I noticed that the attribute specifiers like:
 -pragma-define:CLIB_DEFAULT_SCREEN_MODE=0x3e
..are kept it the __zx_screenmode variable.  I'm masking it out with "and 7" but I noticed doesn't happen elsewhere in zx_common.

zx70, 2024-06-04:
Faster putsprite() on MODE0 and MODE6.
It is an halfway integration, stencil_render() can probably donate the whole colour and video addressing support for all the video modes.

zx70, 2024-07-11:
The latest pull merges the "next column" code originally in stencil_render() reusing it on w_putsprite.

The subroutines mention "MODE6", which is not totally right, but it happened because I'm splitting the improvement in small tasks.  At this stage we have two parallel subroutines to move to the next column in the video memory .

w_putsprite with the introduction of color support is now has an out of range problem on the rightmost edge when in mode0, probably the loops need to be trimmed