z88dk/z88dk #1275: (sccz80) Gameboy z80 support

suborb, 2019-09-10:
Just leaving this here for thoughts. We've managed to support the 8080 without too much trouble, so lets turn to the gbz80:

Missing:

- No index registers
- No alternate registers
- No block instructions
- No in/out (memory mapped io instead)
- No 0xed instruction page (so `ld rr,(nnnn), ld (nnnn),rr, sbc hl,rr, adc hl,rr` are absent)
- No `ld hl,(nnnn), ld (nnnn),hl`
- No `ex (sp),hl, ex de,hl`
- No `djnz`
- No sign or parity flags

Extra (useful) instructions:

- `ld hl,sp+dd`
- `add sp,dd`
- `ld a,(hl+)`
- `ld (hl+),a`
- `ld a,(hl-)`
- `ld (hl-),a`

Movements:

- Opcodes moved: `reti, ld a,(nnnn), ld (nnnn),a`

So it's mostly 8080, the main issues at first glance are the missing `ex` instructions and `ld (nnnn),hl`. `ld hl,(nnnn)` is fairly easy to emulate if we can trash `a`.

pauloscustodio, 2019-09-10:
Shall we use -mgbz80 as the machine specifier option? (to be the same in z80asm and sccz80)

suborb, 2019-09-11:
Thanks for picking up the z80asm side Paulo. 

That’s the value that I’ve chosen - matches the one used by sdcc.

suborb, 2019-09-16:
@pauloscustodio I think djnz has an off by one issue:

Source:

```
flags_again:
        push    hl              ;save fmt
        ld      b,5
        ld      hl,flags
flag_loop:
        cp      (hl)
        inc     hl
        jr      nz,no_flag
; We've found a flag
        ld      a,(hl)          ;pick up flags
        or      c
        ld      c,a
        pop     hl              ;get fmt back
        ld      a,(hl)          ;pick up next character
        inc     hl
        jr      flags_again
flags:
        defb    '-', 0x01
        defb    '+', 0x02
        defb    ' ', 0x08
        defb    '#', 0x10
        defb    '0', 0x04
no_flag:
        inc     hl
        djnz    flag_loop
```

Assembled to:
```
flags_again:
                    push    hl                              ;[0155] e5
                    ld      b,$05                           ;[0156] 06 05
                    ld      hl,$0167                        ;[0158] 21 67 01
flag_loop:
                    cp      (hl)                            ;[015b] be
                    inc     hl                              ;[015c] 23
                    jr      nz,no_flag                      ;[015d] 20 12
                    ld      a,(hl)                          ;[015f] 7e
                    or      c                               ;[0160] b1
                    ld      c,a                             ;[0161] 4f
                    pop     hl                              ;[0162] e1
                    ld      a,(hl)                          ;[0163] 7e
                    inc     hl                              ;[0164] 23
                    jr      flags_again                     ;[0165] 18 ee
flags:
                    dec     l                               ;[0167] 2d
                    ld      bc,$022b                        ;[0168] 01 2b 02
                    jr      nz,$0175                        ;[016b] 20 08
                    inc     hl                              ;[016d] 23
                    stop                                    ;[016e] 10
                    jr      nc,$0175                        ;[016f] 30 04
no_flag:
                    inc     hl                              ;[0171] 23
                    dec     b                               ;[0172] 05
                    jr      nz,$015c                        ;[0173] 20 e7
```

suborb, 2019-09-16:
Status update:

The basics are working:

- [x] sccz80 generates the correct I/O code for 8 bit __sfr
- [x] sccz80 uses the new stack routines
- [x] +test has a -clib=gbz80 target with pretty much the same library as an 8080 port. 
- [x] A fair amount of the test-suite is running 
- [x] z80asm does emulation for ldir and friends

Remaining:

- [ ] A Gameboy crt0
- [ ] Import the standard GBDK library
- [ ] Example
- [ ] Figure out sdcc interop will work (sdcc uses reversed de+hl)
- [ ] sdcc crt0 routines

suborb, 2019-09-21:
I've now got galaxy.c from gbdk compiling and mostly working.

I think sdcc interop goes away at this stage, getting a whole bunch of errors along the lines of:

```
Function canAssign not initialized in port structure
Function notUsed not initialized in port structure
Caught signal 11: SIGSEGV
```

Which will need to be investigated later.

suborb, 2019-09-28:
Closing, will reopen as a separate issue with remaining items

closed 2019-09-28 by suborb