z88dk/z88dk #1276: (z80asm) Gameboy z80 support (see #1275)

pauloscustodio, 2019-09-10:
Update z80asm to handle Gameboy z80 (see #1275)

suborb, 2019-09-13:
@pauloscustodio - thanks for this. Is there a reason why the block instructions aren't replaced with subroutine calls?

pauloscustodio, 2019-09-14:
Yes, the GameBoy does not have ex (sp),hl, which is needed by most block instruction emulators (src/z80asm/dev/z80asm_lib).

I can add emulation code for the ex instructions if the library code can have a scratch pad area of memory to save register values. Can it be in section data_crt?

suborb, 2019-09-14:
Good point. I think a lot of the behaviour that was used for was about making the flag behaviour the same? Given that the gbz80 doesn’t have those flags we can probably simplify and rewrite. 

Having done the timings for ticks I’m not sure that automatically emulating ex de,hl is the right thing to do - it replaces 4 clocks with 36 - so a cheap operation becomes quite expensive.

pauloscustodio, 2019-09-14:
> OTOH, it's very convenient at this stage

I agree. It simplifies porting the libraries a lot.

suborb, 2019-09-14:
Porting the libraries is going quite well since most of the heavy lifting was done during the 8080 porting, re-enabling the block instructions will sort out strings.

There's a few occurrences of ex (sp),hl that are a bit gnarly to untangle. So if we can add a polyfill then life becomes easier (if slower - 168 clocks vs 19), I think the following is correct:

```
        push    af              ;16
        push    de              ;16
        ld      d,h             ;4
        ld      e,l             ;4
        ld      hl,sp+8         ;12, (sp)
        ld      a,(hl)          ;8
        ld      (hl),e          ;8
        ld      e,a             ;4
        inc     hl              ;8
        ld      a,(hl)          ;8
        ld      (hl),d          ;8
        ld      h,a             ;4
        ld      l,e             ;4
        pop     de              ;12
        pop     af              ;12
        ret                     ;16
```

pauloscustodio, 2019-09-16:
Closing this issue. @suborb , please ping if something else is needed for gbz80.

pauloscustodio, 2019-09-16:
> @suborb: I think djnz has an off by one issue

I will have a look

suborb, 2019-10-05:
@pauloscustodio Minor issue that I ran into, the following:

    ld hl,sp-8

doesn't assemble, but needs to be written:

    ld hl,sp+ -8

commit 8aa6f119 (2019-09-13): Fix #1276: add support of GameBoy Z80
commit e322afef (2019-09-17): Fix #1276 off-by-one in gbz80 djnz
commit a1841567 (2019-10-06): Fix #1276: allow a negative offset

Allow ```ld hl, sp-4``` and ```ld hl, sp```
closed 2019-10-06 by pauloscustodio