z88dk/z88dk #1281: New optimization rule (was #1277)

zx70, 2019-09-17:
I wrongly used another issue for this new rule, so here is a dedicated one.

First question: is it correct to get "negative" results for the 8080 related fails ?
The tests are contemporairly declared as "passed"

zcc +test -vn -clib=8080  -I../../framework   -DNO_LOG_RUNNING -DNO_LOG_PASSED ../../framework/test.c main.c strcmp.c stricmp.c strlcat.c strlcpy.c strncmp.c strrev.c strrstr.c strstr.c -o test_8080.bin -lndos  -m
 12:08:52  Starting 8080 test
Starting suite Strcmp Tests (3 tests)
3 run, 3 passed, 655353 failed
Starting suite Stricmp Tests (5 tests)
5 run, 5 passed, 655355 failed
Starting suite Strcasecmp Tests (5 tests)
5 run, 5 passed, 655355 failed
Starting suite Strncmp Tests (4 tests)
4 run, 4 passed, 655354 failed
Starting suite strstr Tests (6 tests)
6 run, 6 passed, 655356 failed
Starting suite strrstr Tests (6 tests)
6 run, 6 passed, 655356 failed
Starting suite Strlcpy Tests (2 tests)
2 run, 2 passed, 655352 failed
Starting suite Strlcat Tests (3 tests)
3 run, 3 passed, 655353 failed
Starting suite strrev Tests (5 tests)
5 run, 5 passed, 655355 failed

Ticks: 2457993

suborb, 2019-09-17:
I've fixed that weird number in my gbz80 branch - I saw it there and then spotted it was in 8080 as well.

At the moment I can't pin down the exact change that fixed it, but don't worry about it at the moment! ;)

zx70, 2019-09-18:
Good, I'm able to run (part of) the test, but I'm now in doubt on the results meaning:
Ticks: 987920		was: 987920
Ticks: 592644		was: 590694 *
Ticks: 2457993		was: 2458032 *
Ticks: 1349599		was: 1349599
Ticks: 1357475		was: 1357475
Ticks: 1336660		was: 1336660
Ticks: 1337737		was: 1337737
Ticks: 1325968		was: 1325968
Ticks: 1351608		was: 1351608
Ticks: 1341013		was: 1341013
Ticks: 732029		was: 706429 *
Ticks: 735860		was: 710260 *
Ticks: 725247		was: 699647 *
Ticks: 725852		was: 700252 *
Ticks: 718073		was: 692473 *
Ticks: 732706		was: 707106 *
Ticks: 728034		was: 702434 *
(...)

A minimal speed increase was expected (even if I'm trying to reduce size), but somewhere ticks are longer.  Is it an exact tick count from an the emulated CPU ?
Is there a way to compare sizes ?
What then ?

suborb, 2019-09-18:
```
        add     hl,sp
        call    l_gint  ;
        add     hl,hl
        pop     de
        add     hl,de
        call    l_gint  ;

zx70, 2019-09-18:
You're probably right  thanks for the hint

zx70, 2019-09-19:
I tried to compare the generated asm files (e.g. by adding '-a' in the make loop of the test suites), but the sources seemed identical.

Thus I focused on the rule effectiveness on my test programs while keeping an eye on the Ticks results.
This rule is more restrictive and is not harming the test suite anymore:

	ld	hl,%1
	push	hl
	ld	hl,%2
	add	hl,sp
	call	l_gint	;
	add	hl,hl
	pop	de
	add	hl,de
	call	l_gint	;
=
	ld	hl,%1
	push	hl
	ld	hl,%2
	call	l_gintsp2de	;

I'm not 100% sure the results will be identical, though.. libraries or something else could take benefit of this new optimization.
I also imagine there's a place where to update the 'Ticks' results for a last comparison ?
Could you (@suborb) please assist in this last step ?

feilipu, 2019-09-19:
Travis is failing, so I added the blocking line back.
```
diff -w tmp2.opt results/Issue_452_arrays.opt
60,61c60
< 	ld	hl,0	;const
< 	push	hl
---
> 	ld	de,0	;const
62a62
> 	push	de
64,65c64,68
< 	ld	hl,2	;const
< 	call	l_gintsp2de	;
---
> 	ex	de,hl
> 	add	hl,hl
> 	pop	de
> 	add	hl,de
> 	call	l_gint	;
Makefile:36: recipe for target 'Issue_452_arrays.opt' failed
make[1]: *** [Issue_452_arrays.opt] Error 1
make[1]: Leaving directory '/home/travis/build/z88dk/z88dk/testsuite'
Makefile:168: recipe for target 'testsuite' failed
make: *** [testsuite] Error 2
The command "./build.sh -e -t" exited with 2.
```

zx70, 2019-09-27:
I put here also an idea for z80rules.1:

	ld	bc,%1
	push	bc
	add	hl,hl
	pop	de
	add	hl,de
=
	ld	de,%1
	add	hl,hl
	add	hl,de

lzdecode size gain:   44468 -> 44450
lzencode size gain:  47650 -> 47632
clisp size gain:  30752 -> 30742

l_gintsp2de in turn depends much more on the way the program was written, the gain in clisp,c is 2 bytes only, but raises to more than 200 bytes in lzencode and lzdecode

zx70, 2019-09-27:
I'm not proud enough to touch those rules againg, though..  :(

zx70, 2019-10-02:
Bad english, I meant: I'm not bold enough  :)

Another possibly useful option (very low frequent case and leaves different flags status):
	xor	a
	ld	(hl),a
	inc	hl
	ld	(hl),a
=
	ld	(hl),0

zx70, 2019-10-13:
let me drop another couple of ideas in this topic:
I decided to make my hand optimization of clisp available (as an 'example' of what shouldn't be done :D).
Most of the custom optimizations are.. just lucky guesses, or worse.. but I think that the l_case_squeezed and others might suggest something clever.  
In example,  a smart optimizer could look for identical code endings (up to the RET instruction) , keep a single one, and jump to it everywhere possible.

obiwanjacobi, 2019-10-14:
The whole problem with how I understand the compiler/assembler works, is that there is no (intermediate) model to reason about the code being generated. That is why you've nothing left but peep-hole optimizations that will never be as effective as something you could achieve with an architecture like LLVM.
Different optimizations are best performed on models of different abstractions. Constant-folding for example is usually done on the AST, while register allocation is done just before going to asm. LLVM has an intermediate model that allows you to reuse existing optimizations (that run on that intermediate model). You can mix and match rules for your purpose and write your own of course.
The Z80 is not a perfect match for the type of processors LLVM was created for, still I think that it could provide better output (generated code) than with any peep-hole compiler, simply because you have more context of how the code interacts.

[2c]
Sorry if its a bit off-topic ;-)

zx70, 2019-10-14:
Yes, at compiler/library level there is plenty of space for improvements, but something is available already, e.g.  I didn't fully rely on the new "__smallc __z88dk_callee" decoration, but having optionally long types in the way I was a bit scared.
Yet a peephole optimizer can, in some case, be more effective, for the same reasons letting rarely an RLE compression win over the Huffman ones  :)

The idea to spot the RET instructions and cut away the identical subroutine endings (if big enough), looks promising.
Good saving will also happen with the JP/JR substitution, when Z80ASM will permit it.

suborb, 2019-10-14:
sccz80 works on a statement by statement basis which means that each individual statement can generate fairly good code for that operation - some of the tricks in `codegen.c` for example. 

What it doesn't have is any context between statements - effectively a statement is:

1. Get values
2. Perform operation
3. Store values

So you end up with sequences of:

1. Store values
2. Get values

The peephole optimiser tried to remove the "get values" bit if it can, I never got quite as far doing this as I hoped to: the semicolons were meant to be used as hints to `copt` to indicate the stack offsets of the parameters.

I've got some ideas about how we can improve things in the compiler, just a question of not being distracted by other things. It's already improved a fair bit over the past few years (frontend and backend) and I think we can squeeze a bit more out of it whilst keeping it running nice and fast.

obiwanjacobi, 2019-10-14:
> The idea to spot the RET instructions and cut away the identical subroutine endings (if big enough), looks promising.

I am sure that could work. I have no idea how frequent this case occurs in code in the wild. Luckily there is a lot of Z80 (C/asm) code out there to try it out on. 
Do you have a model inside the compiler/assembler or is a basically a text search to go through the asm? (have no idea how the peep-hole rules are implemented)

> What it doesn't have is any context between statements

I would think that when you optimize for speed, function inlining and further local optimization could get you a long way? But that's just a gut feeling.

Can the assembler report what rules it ran and how many bytes/cycles were optimized away?
Do you run these rules multiple times, or in a specific order?

zx70, 2019-10-14:
I have nothing at the moment,  just the results of few compiled programs I tried to hand-optimize.   Repetitions in programs are quite common,  but they are harder to intercept even using self modifying code.
A bit of identical tails are yet common e.g. a ret code to be assigned,  the stack equalization etc..

zx70, 2019-10-15:
I checked randomly the programs in "z88dk/examples" and very few cases would gain some byte but the interesting saving possible with clisp.c (I haven't checked them all, though).
By the way very big programs may have huge code portions repeated, e.g. I found in "encode.c" this part repeated 2 times:

	push	hl
	ld	hl,8	;const
	add	hl,sp
	call	l_gint	;
	add	hl,hl
	pop	de
	add	hl,de
	ex	de,hl
	ld	hl,12	;const
	add	hl,sp
	call	l_gint	;
	call	l_pint
	ld	hl,_dad
	push	hl
	ld	hl,14	;const
	add	hl,sp
	call	l_gint	;
	add	hl,hl
	pop	de
	add	hl,de
	ex	de,hl
	ld	hl,6	;const
	add	hl,sp
	call	l_gint	;
	call	l_pint
	pop	bc
	pop	bc
	pop	bc
	pop	bc
	pop	bc
	ret

pauloscustodio, 2019-10-15:
I think this tail-reuse-optimization can be implemented in the z80asm linker, after the data structures passed from the front-end to the back-end are improved, which is also needed to be able to replace JR by JP if the distance is too far. I'll add it to the wish list. (https://github.com/z88dk/z80asm/wiki)

aralbrec, 2019-10-19:
> The Z80 is not a perfect match for the type of processors LLVM was created for, still I think that it could provide better output (generated code) than with any peep-hole compiler, simply because you have more context of how the code interacts.

zsdcc does do these things and its peephole optimizer is more than a text substitution - it does reason about what registers are affected by substitutions to qualify them, eg.  It is sorely missing an equivalent operation for flags, however, which makes it not possible to clean up some of the code that is sometimes generated.   zsdcc's code generation is still far from perfect but the smart peepholer cleans it up quite a bit and is the main reason zsdcc generates better code than its parent sdcc.

obiwanjacobi, 2019-10-19:
> There is an experimental compile sequence that has llvm optimizing c code that is then passed through zsdcc to generate code. However, the code quality is lower than using zsdcc directly and that's probably because generic llvm is not geared for targets less than 32 bits.

I don't think using LLVM to generate C code and then compile that with zsdcc is a good strategy. The whole idea of LLVM is to use the intermediate models to implement these optimizations. I am not aware that LLVM has a 32-bit assumption. I am just starting to read about LLVM but what I understand is that the types are initially decided by the front end. So if C++ defaults to a 32-bit integer, that is what you will see in the models. If you were to use the stdint types explicitly, these should show up in the models too.

closed 2020-07-06 by suborb