z88dk/z88dk #1282: [mz] Latest snapshots may have broken the mz target

Fabrizio-Caruso, 2019-09-19:
If I try to load the Cross Chase binary for the mz target the emulator now gets in an infinite loop while printing the pi character.

I am testing this with the emulator mz800emu-1.0.3-win32
which I had previously used successfully to test Cross Chase

The mz target is built (as previously) with:
```
mz:
	$(Z88DK_PATH)$(MYZ88DK) +mz $(SCCZ80_OPTS) \
	-D__MZ__ -clib=ansi -pragma-define:ansicolumns=32 -vn \
	-DFULL_GAME -DSOUNDS  -DEND_SCREEN -DBETWEEN_LEVEL \
	-DCONIO -DNO_INIT_GRAPHICS \
	-lndos -create-app -o $(BUILD_PATH)/FULL_mz.prg \
	$(SOURCE_PATH)/end_screen.c \
	$(FULL_FILES)
	rm $(BUILD_PATH)/FULL_mz.prg
	mv $(BUILD_PATH)/FULL_mz.mzt $(BUILD_PATH)/FULL_mz.mzf
```
[image]

zx70, 2019-09-20:
I may have not a perfectly clean z88dk setup but I seem to have problems on all the targets and different applications.
Still I can't understand the root cause, few first attempts to revert my recent changes didn't solve.

zx70, 2019-09-20:
Not as serious as it looked, I incidently checked this issue on othello.c, on which I put a peculiar optimization 9 months ago (specified callee/fastcall header declarations for C functions within the game) , now crashing.
Did you do the same in your program ?

zx70, 2019-09-20:
More detail on what I noticed while hunting for your bug:  I was forcing the __z88dk_callee mode for funtions requiring recursion.   It may be that now SCCZ80 is more effectife on such directives than 9 months ago, so the program behavior was compromised by my optimization.
I'm not expecting Fabrizio is having the same problem, but he worked had on optimizations, so you never know  :)

Fabrizio-Caruso, 2019-09-21:
The problem is not fixed by snapshot 20/09/2019.
I compile with no optimization option so I suppose the compiler uses O2:
```
mz:
	$(Z88DK_PATH)$(MYZ88DK) +mz $(SCCZ80_OPTS) \
	-D__MZ__ -clib=ansi -pragma-define:ansicolumns=32 -vn \
	-DFULL_GAME -DSOUNDS  -DEND_SCREEN -DBETWEEN_LEVEL \
	-DCONIO -DNO_INIT_GRAPHICS \
	-lndos -create-app -o $(BUILD_PATH)/FULL_mz.prg \
	$(SOURCE_PATH)/end_screen.c \
	$(FULL_FILES)
```
I confirm that Cross Chase works fine with other targets such as C128 (compiled with zsdcc), Spectrum (compiled with sccz80). 
I use default optimization options.

Fabrizio-Caruso, 2019-09-21:
Ok! This issue is now different...
The game eventually loads correctly after a MUCH longer time and after printing zillion PI characters...

Snapshot 2018 09 21 loads the game immediately and prints different random looking characters
Recent snapshots and less recent snapshots such as 2019 01 01 display PI characters and load the game after a much longer time

zx70, 2019-09-22:
It could have been broken also before.  I recently read about stack limits on the mz target.  Are you using the trick described here?
https://www.z88dk.org/forum/viewtopic.php?id=10993

Fabrizio-Caruso, 2019-09-24:
Compiling with:
```
	-pragma-define:CLIB_EXIT_STACK_SIZE=0
```
does not fix the problem

Fabrizio-Caruso, 2019-09-24:
On the other hand I have managed to fix it with a set of pragmas I use to same memory:
https://github.com/Fabrizio-Caruso/CROSS-CHASE/blob/master/src/cross_lib/cfg/z88dk/zpragma_clib.inc

zx70, 2019-09-26:
-pragma-define:REGISTER_SP=0xd000  resolved for Fabrizio.
I'm tempted to force sp to that value,  it could still be overridden in case of reduced ram

zx70, 2019-09-27:
ATM it is in the heading example in the wiki 
Target working

closed 2019-09-27 by zx70