z88dk/z88dk #1285: [zx81][wrx] clg() breaks my game

Fabrizio-Caruso, 2019-09-25:
I have a fully working Cross Chase version for the ZX81 +16K +WRX.

It works fine if I avoid clg(). 
I compile the game with:
```
zx81_16k_wrx_p:
	$(Z88DK_PATH)$(MYZ88DK) +zx81 -subtype=wrx -clib=wrx \
	$(SCCZ80_OPTS) -O3 \
    -DANIMATE_PLAYER \
	-DFORCE_XSIZE=36 \
	-DFORCE_YSIZE=30 \
	-vn \
	-lndos \
...
	-create-app -o  $(BUILD_PATH)/LIGHT_zx81_16k_wrx_p.prg \
	$(LIGHT_FILES) \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/zx81/zx81_init_graphics.c \
	$(CROSS_LIB_PATH)/display/alt_print/memory_mapped_alt_print.c
```
Is there a reasonable substitute?
I am currently just printing spaces... and this is very slow.

Fabrizio-Caruso, 2019-09-25:
This may well be an invalid because I may have used all available memory.

zx70, 2019-09-26:
I confirm it may happen,  clg hits much memory at once.

closed 2019-09-26 by zx70