z88dk/z88dk #1286: [feature request][gamegear] Support for Gamegear (CPU is Z80)

Fabrizio-Caruso, 2019-09-28:
I wonder if there was a technical reason why GameGear is not supported even though Sega Master System has been supported by porting devkitSMS to Z88DK.

It would be call to cover this well-known Z80 system.

zx70, 2019-09-28:
I think the GameGear is kind of supported already being very close to the MasterSystem.
What do you think to be missing?
Extra appmake support?  Docs?  Facilities for shorter Sprite tiles?

Fabrizio-Caruso, 2019-09-29:
I meant some minimal appmake and library support. All the rest would be nice to have.
In the current situation Z88DK is not usable to develop for the GameGear.
I would not be able to use it even if it were just a hello world print.

suborb, 2019-09-29:
Have you read this blog post? http://sebastianmihai.com/main.php?t=119&n=Sega-Game-Gear-development-Burgers-of-Hanoi-GG

It looks like the main things are the screen size and the palette which look like they can be changed with what’s already there.

Fabrizio-Caruso, 2019-09-29:
Thanks! It would be good to a specific target or sub-target that takes care of these minor changes.

zx70, 2019-09-29:
Thank you Fabrizio, this is more specific. 
Did you succeed with the Sega Master System ?
If so let's focus on the differences,  I wonder if the same binaries can be directly used or converted with some existing tool.. I'd bet so.

Fabrizio-Caruso, 2019-09-29:
@zx70, yes, I have succedded in building a (very ugly and minimalistic) Cross Chase version for the SMS long ago

zx70, 2019-09-29:
understood,  probably:
http://www.smspower.org/forums/14742-GameGearToMasterSystemHacks
what I can imagine reading here there might be both input and output issues. 

is the binary format a problem too ?  do the game gear emulators refuse the file.?

zx70, 2019-11-08:
Very basic correction to the text output coordinates. 
Text will still be printed with a different palette but at the expected position.   CLS code isn't still implemented,  neither text scrolling, but gotoxy should work.
Is a 20x16 characters rez enough for your game?

zx70, 2019-11-08:
I forgot the instructions:  just add -lgamegear to your zcc command, using the +sms target.

Fabrizio-Caruso, 2019-11-08:
Thanks! My game could still be playable somehow with a character grid as small as 6x8.

zx70, 2019-11-08:
good.  I managed to make also load_palette behave in the same way in the gamegear and the SMS.
Obviously a new load_palette should be necessary for the extra colour capabilities of the Game Gear.

Fabrizio-Caruso, 2019-11-18:
@zx70 I am going to try to test it with my universal game.
Which emulator, compile options, emulator options should I use?

Fabrizio-Caruso, 2019-11-18:
If I compile with:
```
$(Z88DK_PATH)$(MYZ88DK) +sms -lgamegear -clib=sdcc_iy -startup=17 -v
```
i.e., as I successfully do with the sms target except for `-lgamegear` I get an error about the `gamegear.lib` not found:
```
...
PROCESSING chase/strategy.c
zsdcpp -I"cross_lib" -I"cross_lib/sleep" -I"cross_lib/display" -I"cross_lib/display/graphics_data" -I"cross_lib/display/graphics_data/cc65" -I"cross_lib/display/graphics_data/z88dk" -I"cross_lib/display/graphics_mode" -I"cross_lib/display/alt_print" -I"cross_lib/display/image_settings" -I"cross_lib/display/redefine_characters" -I"cross_lib/include" -I"cross_lib/sound" -I"cross_lib/sound/cc65/atmos" -I"cross_lib/sound/cc65/c264" -I"cross_lib/sound/cc65/pokey" -I"cross_lib/sound/cc65/sid" -I"cross_lib/sound/cc65/vic20" -I"cross_lib/sound/z88dk/bit_bang" -I"cross_lib/sound/z88dk/psg" -I"cross_lib/text" -I"cross_lib/input"  -iquote"." -D__Z88DK -D__SMS -DSMS -D__SMS__ -D__Z80 -DZ80 -DFORCE_XSIZE=20 -DFORCE_YSIZE=16 -D__SMS__ -DNO_COLOR -DTINY_GAME -DSTDLIB -DLESS_TEXT -DBUFFERED Error: cannot read file 'gamegear.lib'
-DTURN_BASED -DALT_PRINT -DALT_CLEAR_SCREEN -DNO_SLEEP -DNO_WAIT -D__SDCC -D__SDCC_IY -isystem"c:/z88dk/\include\_DEVELOPMENT\sdcc"  -D__SDCC  "chase/strategy.c" "C:\cygwin\tmp\zcc14F414.i2"
z88dk-zpragma  < "C:\cygwin\tmp\zcc14F414.i2" > "C:\cygwin\tmp\zcc14F414.i"
zsdcc --constseg rodata_compiler   --reserve-regs-iy -mz80 --no-optsdcc-in-asm --c1mode --emit-externs  --no-c-code-in-asm   --no-peep --peep-file "c:/z88dk//libsrc/_DEVELOPMENT/sdcc_peeph.2"  < "C:\cygwin\tmp\zcc14F414.i" -o "C:\cygwin\tmp\zcc14F414.opt"
z88dk-copt  "c:/z88dk//libsrc/_DEVELOPMENT/sdcc_opt.1" < "C:\cygwin\tmp\zcc14F414.op1" > "C:\cygwin\tmp\zcc14F414.opt"
z88dk-copt  "c:/z88dk//libsrc/_DEVELOPMENT/sdcc_opt.9" < "C:\cygwin\tmp\zcc14F414.opt" > "C:\cygwin\tmp\zcc14F414.op1"
z88dk-copt  "c:/z88dk//libsrc/_DEVELOPMENT/sdcc_opt.2" < "C:\cygwin\tmp\zcc14F414.op1" > "C:\cygwin\tmp\zcc14F414.asm"
type "C:\cygwin\tmp\zcc14F414.asm" >> "C:\cygwin\tmp\zcc14F414.tmp"
type "C:\cygwin\tmp\zcc14F414.tmp" >> "C:\cygwin\tmp\zcc14F414.asm"
z80asm -D__SDCC_IY -I"c:/z88dk/\libsrc\_DEVELOPMENT\target\sms" -D__SDCC  -s  -mz80 -I"c:/z88dk//lib" "C:\cygwin\tmp\zcc14F414.asm"

PROCESSING c:/z88dk/\libsrc\_DEVELOPMENT\target\sms\sms_crt.asm.m4
m4 -I "c:/z88dk/\libsrc\_DEVELOPMENT\target\sms" -I "c:/z88dk//src/m4"  --define=__STARTUP=17  < "c:/z88dk/\libsrc\_DEVELOPMENT\target\sms\sms_crt.asm.m4" > "C:\cygwin\tmp\zcc14F415"
type "C:\cygwin\tmp\zcc14F415" >> "c:/z88dk/\libsrc\_DEVELOPMENT\target\sms\sms_crt.asm"
type "c:/z88dk/\libsrc\_DEVELOPMENT\target\sms\sms_crt.asm" >> "C:\cygwin\tmp\zcc14F415.asm"
type "C:\cygwin\tmp\zcc14F415.asm" >> "C:\cygwin\tmp\zcc14F415.tmp"
type "C:\cygwin\tmp\zcc14F415.tmp" >> "C:\cygwin\tmp\zcc14F415.asm"
WILL ACT AS CRT

GENERATING OUTPUT

PROCESSING CRT
z80asm -D__SDCC_IY -I"c:/z88dk/\libsrc\_DEVELOPMENT\target\sms" -D__SDCC  -s  -mz80 -I"c:/z88dk//lib" -I"c:/z88dk/\libsrc\_DEVELOPMENT\target\sms"  "C:\cygwin\tmp\zcc14F415.asm"

z80asm -mz80 -b -d  -o"../build/TINY_gamegear_buffered.bin" -m -s -L.  -L"c:/z88dk/\libsrc\_DEVELOPMENT\lib\sdcc_iy"   -D__SDCC_IY -I"c:/z88dk/\libsrc\_DEVELOPMENT\target\sms" -D__SDCC -igamegear -isms    "@C:\cygwin\tmp\zcc14F416.lst"
make: *** [Makefile:7520: gamegear_tiny_buffered] Error 1

```

zx70, 2019-11-19:
https://github.com/z88dk/z88dk/wiki/Platform---SMS

It is related to the classic library only,  in the new one the functions are not yet available,  I'm sorry I didn't remember the differences

Fabrizio-Caruso, 2020-03-28:
Any progress to report? Should I rather use classic lib routines Cross Chase abstraction layer?

aralbrec, 2020-04-15:
devkitsms itself has the option to be built for the gamegear where it changes some of the functions.  I didn't know what the differences were exactly between the gamegear and sms as I only had technical information for the sms so I decided to leave it for another time and perhaps for a separate target.The devkitsms version in z88dk is an asm version of the generally c

closed 2021-06-04 by Fabrizio-Caruso