z88dk/z88dk #1288: yaz180 - subtype=app -clib=new -> incorrect program exit

feilipu, 2019-09-30:
An old issue, that I've avoided for some time. But I've run out of ideas.

In the yaz180 platform with `-subtype=app -clib=new`, the program should jump to the monitor by calling the ['_exit_far'](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/yaz180/crt_config.inc#L99) function in common memory on exit. Indeed, with sdcc this is what happens normally.

But with a sccz80 `-clib=new` compile a very simple "Hello World" program exits to a reboot when any `printf()` statement is included in the program. The exit to reboot only happens if there is a `printf()`. Otherwise, it exits correctly back to the monitor, and this can only happen via a call to `_exit_far`.

Could that mean that there is something unbalanced in the sccz80 `printf` routines, or exit strategy?

I can't see any differences in the listings for the two options, except for one stack thing in `vfprintf_unlocked` that is different, where there are 4 of occurrences of [`dec bc` pairs](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/stdio/z80/asm_vfprintf_unlocked.asm#L1061) which seem to differ, between the two compilers.

Any thoughts on what might be the cause?

suborb, 2019-09-30:
The `inc/dec` will be to cover the the different (i.e. R->L and L->R) argument passing.

I'm guessing the compiled code looks sensible for the hello world case?

Do the statements following the `printf()` work correctly? What happens if the `printf()` is put into a function? Does that function return and execution continue as expected?

feilipu, 2019-09-30:
Yes. the program seems to execute correctly (when it is more complex), and the "Hello World" does exactly what it should too.

The only issue is around the program exit. It should (by all rights) also call the `_exit_far` function, like it does when there is no `printf()` present, but it simply doesn't.

Or, perhaps it doesn't call `_exit_far` correctly? But, I don't really think that is a possibility though, as [`_exit_far`](https://github.com/feilipu/yaz180/blob/master/yabios/asm_common1_driver.asm#L358) doesn't care about the entry stack, it just saves it, switches banks, and recovers the old system stack, with the monitor proceeding where it left off.

suborb, 2019-09-30:
Just checking the basics! It looks like the stack restoration after `printf()` is good then. What about a `puts()` - if that breaks then the it's the IO routines causing problems.

Is there anyway you can print the contents of the system stack whilst it's running? - I'm wondering if that's getting corrupted somehow.

feilipu, 2019-09-30:
I've just done a test with `puts("hello world\n");` and it is not working in the same way too. So, that points to a difference in the way that the two compilers call the I/O routines, somehow, I guess. Or something with the I/O routines themselves.

Not sure of the best way to print the stack on the fly. I can output / display a byte on the 82C55 port and capture the last place a program gets too, and potentially also where it went by setting bits in the latched display byte.

Thought... could the calling of or exit from of the I/O routines be different, and cause this?

feilipu, 2019-10-01:
Hmm... this is quite a curious problem.

I had a theory that the I/O routines, being interrupt driven, were being interrupted by the `__crt_on_exit` running in banked memory, and were returning to an incorrect location on the bank 0 system upon return. Of course, this can't be the case because the interrupt I/O is being run with interrupts disabled.

There is a system_tick run 256 times per second triggered by the PRT0, so potentially this could be the cause too, but again it is being run with interrupts disabled.

So I wrote a small delay routine, to ensure that the I/O is being completed before the jump.

```asm
      ; program jumps to absolute address     
      IF (__crt_on_exit & 0xffff)
        
         xor a
         ld bc,0x4001
         .test_point
         out (c),a        ;output A on 82c55 Port B
         ex (sp),hl
         ex (sp),hl
         ex (sp),hl
         ex (sp),hl
         ex (sp),hl
         ex (sp),hl
         ex (sp),hl
         ex (sp),hl
         halt
         inc a
         jr NZ,test_point

         jp __crt_on_exit & 0xffff
      
      ELSE
      
         rst 0
         
      ENDIF
```
And sure enough the `puts()` output is complete, and I can watch the hex display count up and when it reaches the end with sccz80 it jumps the correct location, but the system is reset. The `halt` was placed to see if interrupts are still enabled at that point. And indeed the timing PRT0 is still firing otherwise the loop would stop.

So I tried inserting a `di` just prior to the `jp __crt_on_exit & 0xffff` to see whether something interrupt driven was causing the issue, but this produced no change in effect.

Now I have a theory that the [`_exit_far`](https://github.com/feilipu/yaz180/blob/master/yabios/asm_common1_driver.asm#L358) routine may be behaving differently depending on what the compiler leaves on the program stack, because it is unbalanced, and is doing something odd.

TBC.

feilipu, 2019-10-03:
Hmm. Further interesting indications.
I made a few tweaks to the ACIA driver, but there was no impact on the symptom. But, in the course of testing the platform I loaded CP/M. In my CP/M I have a CCP [EXIT](https://github.com/feilipu/yaz180/blob/master/cpm22/cpm22.asm#L1214) function, which calls `_exit_far`.

In most situations this EXIT function returns to the bios prompt to allow CP/M to be restarted with different drive files (essentially to change disks).

But, if DDIR is run then the EXIT fails and the machine restarts. What is interesting this? DDIR is compiled with sccz80, and it should only have access to the same I/O routines as the rest of CP/M, but somehow it is causing the same symptom as other "native" functions compiled with CP/M.

This leads me to think that somehow the system stack is being overwritten by one thing or another. DDIR is sccz80/classic library. Native functions are sccz80/new library. Now to find out why exactly.

feilipu, 2019-10-04:
4 nights on this. Can't see where things are going wrong, or even what is going wrong.
Closing until I can bare to look at this again.

closed 2019-10-04 by feilipu

zx70, 2019-10-04:
Today I was debugging a faulty fgets() function oddly masking the whole test program running twice,  the first time correctly picking the string from a file and printing it,  the second time becoming user interactive and using the console. 
It took me some time to discover it was caused by the string buffer overflow..