z88dk/z88dk #1290: (z80asm) Linker support to banked memory

pauloscustodio, 2019-10-03:
In issue #1287 two options were presented for banked memory support in z80asm. Option 1, which is independent of any banked memory architecture, was implemented.

This issue collects the discussion for a linker support for banked memory, to allow development for a large code area model with code overlays. This proposal simplifies Option 2 by committing to a memory model similar to the one used by the Rabbit with 24-bit address range and solves the issue of wrong stack offsets in Option 2.

Assumptions:

- Data pointers are 16-bit and data memory is not banked. This matches the Rabbit memory model and keeps the compiler infrastructure for using 16-bit addresses.

- Near code pointers are 16-bit and point to an address range that is always paged in. This allows calls to the most common functions (e.g. the C library) to be simple CALL statements. This matches the Rabbit memory model, where the address range 0x000-0xDFFF is always paged in

- Far code pointers are 24-bit and a fixed 16-bit address range is used to access all banked code memory. This matches the Rabbit memory model, where the address range 0xE000-0xFFFF is a window where the 24-bit code addresses can be mapped by setting the XPC 8-bit register.

- The z88dk library implements a platform-dependent "trampoline" function that is used to call far pointers and keeps track of the last paged-in bank.

Possible implementation:

- z80asm defines a new BANKED directive to define a new section that will collect all the far code.

    BANKED [start-addr,size]    ; use instead of "section xxx" for banked code, parameters optional

- z80asm defines a new command line option to receive the banked memory window range, as this is platform dependent and needs to be supplied by the crts; e.g. --banked=0xE000,0x2000	

- The linker collects all banked modules and assigns the 24-bit addresses, generating one binary file per bank (BANK_hh.bin). The total code size per module (i.e. source file) in the banked area needs to fit in the bank size. This is to allow the linker to collect modules without splitting one module across banks. The linker will collect modules by size, from larger to smaller, to attempt to fill each bank as much as possible and minimize the number of banks.

- z80asm defines a new BCALL directive to implement far-calls, but that can also be used for near calls (with some waste). It is assembled and patched at link time according to the following rules:

    - if the target address is greater that 0xFFFF and the lower 16-bits are not in the bank memory window, issue an error;
	  
    - if the target address is outside the bank memory widow, the linker patches a CALL instruction, followed by NOPs;
	
	- if the target address is inside the bank memory window and in the same bank, the linker patches a CALL instruction, followed by NOPs;
	
	- if the target address is inside the bank memory window and in a different bank, the linker patches a CALL instruction to the trampoline function, followed by the 24-bit target address; the trampoline function collects the 24-bit target address, switches memory if the bank id is different from the last called address, and jumps to the target address.

- The C compiler needs a new function attribute (__banked?), collects the banked code in a BANKED section and replaces calls to banked functions by BCALL.

- The crts define the trampoline function and the --banked= option to z80asm for each architecture.

suborb, 2019-10-03:
Thanks @pauloscustodio that's a lot to digest (which I haven't yet). Auto populating banks will make a lot of people very happy.

I know @EtchedPixels isn't a user, but I think his feedback based on experience with Fuzix on different targets would be valuable.

obiwanjacobi, 2019-10-03:
Sorry to butt-in... :-)

Am I correct in saying that this assumes that there is (only) one memory region where extra memory is swapped (banked) into? As in the command line example `--banked=0xE000,0x2000` - a 'window' if you will, at address 0xE000 of 0x2000 bytes in size. That would also mean that all code in banks will be targeted at that address (not relocatable)...? Or could there be more than one 'window'?

I have not digested this fully yet, but the linker scraping together code to fill banks would be a really useful feature. Regardless from any further decisions on how to implement banking, that feature alone would be worth having - and should be usable as a stand-alone feature to facilitate other banking schemes (at least somewhat). You also want to be able to generate relocatable code for those banks.

There is already an interface for accessing far memory (for C https://github.com/z88dk/z88dk/wiki/More-Than-64k). Could this not be reused (for calling a far function)? 

[back to digesting this]

pauloscustodio, 2019-10-03:
> Sorry to butt-in... :-)

You're welcome!

> Am I correct in saying that this assumes that there is (only) one memory region where extra memory is swapped (banked) into?

Yes, that was the idea, based on the Rabbit architecture that has the address space $0000-$DFFF always fixed and the window $E000-$FFFF is used to map any of the 1Mb address space, by changing the XPC 8-bit register.

>  That would also mean that all code in banks will be targeted at that address (not relocatable)...? 

Yes, that was the idea. The linker would allocate fixed addresses in that range to all the banked code and patch in far-calls for all calls to __banked functions.

> You also want to be able to generate relocatable code for those banks.

I think you mean at load-time, i.e. when the code is loaded from external storage to memory, the fixed addresses my need to be patched. If a bank is not shared between processes, then only the bank id needs to be patched. Any suggestion on the interface between the linker and the loader?

> There is already an interface for accessing far memory

Thanks for the reference, I'll have a look.

ped7g, 2019-10-03:
(JFYI) The ZX Spectrum Next allows to remap whole 64kiB address space of Z80N, per 8kiB block.

EtchedPixels, 2019-10-03:
I already implemented the needed compiler support., and a hacky version of the linker support for Fuzix (see my git tree for SDCC patches and the linker mods live in the Fuzix tree). The sdcc patch is tiny and the linker patch likewise. It's a bit hacky because I didn't want to get into the bigger changes to automate segment placing, or reading an overlay file description table (the way the native CP/M tools did)

Basically I generate

push af
call foo
pop af

for all function calls, which means that the arguments are always at the larger offset so there is no messing around with painful crap like 'far' and 'near' functions. I had too much of that in the DOS world 8) It also means it works with asm->C and C->asm which is kind of handy in an OS.

The linker switches any push af call pop af sequences that go between banks into

    call __bank_0_1
    defw function

and the bank hooks do a bit of stack gymnastics so they can stack the previous bank in the extra word whilst returning via a second stub that switches bank back so that I can call arbitrarily between banks, between common and banks and even do context switches and interrupts this way.

For address references to functions (eg function tables) I generate a short stub in common space for each reference. Stubs to the same function are shared. That in turn means that function pointers are still 16bit _and_ the correct C semantics for pointer comparison are maintained.

No 32bit adresses, no 24bit pointers (both of which kill you especially as SDCC is relatively poor at 32bit operations), no __far garbage and the same code is trivially compiled for banked and non banked systems.

Feel free to lift chunks of it as you want.

@suborb actually I do use z88dk - I wrote the Amstrad NC support 8). I've not added Fuzix support into Z88DK because my intent is to redesign the syscall interface and binary format format (hopefully this winter) and I wanted to leave Z88DK work until after that.

obiwanjacobi, 2019-10-04:
@pauloscustodio 
Yes, I do mean re-address at load-time. If I remember correctly z88dk already supports that - don't know the details of that support though.

> Any suggestion on the interface between the linker and the loader?

I have some notes on some thoughts on loading programs into my 4k-page banking system (1MB max) - any page can popup anywhere (except page0). Reading it back I am uncertain it is the right way to go. Basically I envisioned the program having a header with some extra meta data that would tell the loader what to do. _**After the program is loaded the actual header could then be discarded.**_ Because my memory-pages are relatively small I foresaw that I would need some way to keep memory pages together -groups of 4k-pages that always would be swapped together.
Not sure how useful these details are for a general case - I do not see many designs that use the granularity I have.

Another thing I considered was what should happen if this program was a module (library) and had dependencies on other modules. That is another can of worms ;-)

> If a bank is not shared between processes

Are you considering that / are there any targets that actually support (isolated) processes!?
Copy-on-write springs to mind...

@EtchedPixels 
> which means that the arguments are always at the larger offset

How does that work? `<over-my-head>`

EtchedPixels, 2019-10-05:
@obiwanjacobi  There are two cases

1. A local call which executes

push af
call foo
pop af

meaning the arguments are at SP+4 upwards and the stack is

return address, nonsense, arg, arg, arg, ...

and

2. A banked call in which case it is

call __bank_1_2
defw addr

and the stack when the __bank_1_2 code has run and called addr is

return addr   bank_2_1_return args ..

(where bank_2_1_return is code that switches to bank 1 and returns to the return addr proper)

So the arguments are in the same place and there are no "near" or "far" functions

The compiler patch is simple. It recognizes a new argument, it puts push af/pop af around function calls and it uses 4 not 2 as the argument base.

Relocation on swap in is a lot harder than you think by the way: if you swap something in and it does a call, you cannot then swap it out and swap it back in because your return address and potentially preserved register values are now wrong and you might also return into it without a chance to catch the fact. It isn't impossible but it's very hard - you might want to read up on how Windows 16bit worked, and then when you've finished feeling ill reconsider 8)

https://devblogs.microsoft.com/oldnewthing/20120622-00/?p=7303
https://devblogs.microsoft.com/oldnewthing/?p=11203

but note that it's a bit uglier for Z80 because you don't have adjustable segment bases and always push and set up IX. Rabbit is kind of OK because all your banks are in the same top window but on Rabbit you really don't want to go via IX anyway.

In the perfect world by the way there is also a use case for data in banked areas. If the code does not take the address of a static object the linker can place it in the same bank as the code that references it since there is no valid C way of referencing the object from another module. If the address is ever taken however then it has to be in common space. That's something I've not bothered tackling directly as for the Fuzix case it's rare.

Copy on write doesn't matter unless you support Z280 (which needs a really quite different compiler for the best results). Z80 and Z180 can't do fault/restart even with an external MMU. You can do shared code by mapping the same bank into multiple processes if your memory layout and banking is suitable but you can't do COW.

I've got the Fuzix kernel code running in 2 x 32K banks, and in 4 x 16K banks (ZX spectrum 128K and Tom's SBC for example) with the SDCC and linker patches I have. Works very nicely.

obiwanjacobi, 2019-10-07:
The stack part I get. Thanks for explaining. 

Relocation on swap is not what I meant. I meant relocation on load - as in adjusting the base address for the code block just loaded into a bank.

More thoughts on this are that it seems a waste not to use banks for self-contained data. I am thinking of graphics for games, tile maps. Any thing self-contained. See Relative Pointer in my doc https://docs.google.com/document/d/1tF7p2I6861tk6l0KzyiMkYkhx_dNrJLCjcLoVhHxSHA/edit?usp=sharing

Although you can make banked calls inside banked code, you will need a locking mechanism for data. When the program has access to a real 16-bits pointer, that bank 'window' is locked untill it is explicitly released. 

That brings me to my second point: support more than one bank window. The physical location of these windows should not be in code but in the target config. A bit like how the memory map (sections) works now..?
Having more than one bank window also requires the window to be encoded into the bank Id that is put on the stack to revert back to the previous thank-you have to know what window to change. 
Putting the bankId into the stack has consequences for the stack offset to use for parameters. 

My 2c's

EtchedPixels, 2019-10-07:
@obiwanjacobi If you have that kind of data and use case then yes  but it's surprisingly hard to use in C without writing very nasty OS specific code. Your design reads like a subset of Windows 16bit for Z80 with all the same requirements about having to lock things to prevent them moving to make room for stuff. Actually in some ways it's less bad because your don't have variable sized blocks as badly as Win16 did with segments so your requirement to move stuff to make holes shouldn't be as bad.

Depends what you are trying to do. My goal was to run normal ANSI/POSIX/Unix C and provide a standards compliant behaviour set.