z88dk/z88dk #1296: [msx][bug] mode1 broken? VDP remapped?

Fabrizio-Caruso, 2019-10-08:
My game Cross Chase no longer works.  My game uses mode 1 and writes directly into video ram. Has the VDP been remapped in the recent weeks/months?

Moreover it seems that mode 1 and conio no longer work together.

Shouldn't `vpoke(0x1800,65)` display an 'A' on the top left corner of the screen?

Nothing is visible on the screen:
```
#include <msx/gfx.h>
#include <conio.h>
int main()
{
    set_color(15, 1, 1);	    
    set_mode(mode_1);   
    gotoxy(2,2);
    cprintf("Hello world");
    vpoke(0x1800,65);
    while(1){};
} 
```

suborb, 2019-10-08:
I presume mode 1 means MSX SCREEN 1 so VDP mode 0 - 32x24?

The VDP addresses were harmonised across all targets at the start of the year, details on the wiki here: https://github.com/z88dk/z88dk/wiki/Classic-TMS9918

Fabrizio-Caruso, 2019-10-08:
@suborb, then there seems to be a bug because nothing is shown on the screen.
Shouldn't `vpoke(0x1800,65)` display something on the screen?

suborb, 2019-10-08:
Try `vpoke(0,65)` - the name table is at address 0x0000

Fabrizio-Caruso, 2019-10-08:
@suborb I get nothing on the screen.
Even the following code produces nothing:
```
#include <msx/gfx.h>
int main()
{
    unsigned char i;
    set_color(15, 1, 1);	    
    set_mode(mode_1);   
    for(i=0;i<254;++i)
    {
        vpoke(0x0000+i,i);
    }
}
```

`set_mode(mode_1)` is no longer working. It should set text mode 32x24.

Fabrizio-Caruso, 2019-10-08:
@suborb, @zx70 
I should get the following picture, which I do get with a previous version of Z88DK (and using 0x1800 as base address):
[image]

Fabrizio-Caruso, 2019-10-09:
@suborb 
I am compiling with:
```
	$(Z88DK_PATH)$(MYZ88DK) +msx $(SCCZ80_OPTS) -lmsxbios -vn -lndos \
	$(SOURCE_PATH)/../test/msx_test.c \
	-create-app -o $(BUILD_PATH)/TEST_msx.prg
``` 
Maybe the problem is in the video ram. It could be that rom characters are not copied there now.
Could you please provide a hello world example in text mode 32x24 and the compile instructions?
It is not clear to me how I should adapt my code to the latest Z88DK.

Fabrizio-Caruso, 2019-10-09:
I got it! mode_0 is mode_1 now!

suborb, 2019-10-09:
Sorry, sleep/work got in the way!

The preferred way of printing characters on the screen is with `--generic-console` using the z88dk drivers (these are in libsrc/video/tms9918). Using this subsystem the screen mode numbers are those in the VDP datasheet and the screen offsets are as on the wiki page. 

However, we've also got the `<msx.h>` library, which comes in two flavours on the MSX:

1. Generic - hits the hardware directly and uses the modes and offsets documented above
2. `-lmsxbios` - this uses the MSX firmware

In case 1, I've just committed some code to set the font when the mode is switched. Previously there was nothing loaded into the pattern table so the screen was gloriously blank (which I think is what you saw). The font can be added at compile time via `-pragma-redirect:CRT_FONT=....`

`-lmsxbios` uses the screen mode numbers, offsets and font defined by the MSX firmware - so it will use the MSX font and the normal MSX offsets.

So using the generic libraries, the following will work:

```
zcc +msx -subtype=rom mode.c -create-app
#include <msx/gfx.h>
int main()
{
    unsigned char i;
    set_mode(mode_0);
    for(i=0;i<254;++i)
    {
        vpoke(0x0000+i,i);
    }
    while(1);
    return 0;
}
```

Using -lmsxbios, the following will generate the same result:

```
zcc +msx -subtype=rom mode.c -create-app -lmsxbios
#include <msx/gfx.h>
int main()
{
    unsigned char i;
    set_mode(mode_1);
    for(i=0;i<254;++i)
    {
        vpoke(0x1800+i,i);
    }
    while(1);
    return 0;
}
```

There's a previous issue #1214 describing this switch mode number switch round. It's confusing and I'm not sure what the right answer is.

Fabrizio-Caruso, 2019-10-09:
So, if I understand well, tomorrow's snapshot fixes the lack for rom fonts? 
With the current latest snapshot I confirm that the following code prints nothing when compiled with `... +msx -subtype=rom` does not work:
```
#include <msx/gfx.h>
int main()
{
    unsigned char i;
    set_color(15, 1, 1);	    
    set_mode(mode_0);
    for(i=0;i<254;++i)
    {
        vpoke(0x0000+i,i);
    }    
    while(1);
    return 0;
}
```

Fabrizio-Caruso, 2019-10-09:
@suborb I have just opened the new issue #1297 the may be related to this one.

suborb, 2019-10-11:
I've just merged #1300 which should resolve this issue by restoring the mappings to the standard MSX mappings. Please reopen if it's not the case.

closed 2019-10-11 by suborb