z88dk/z88dk #1297: [msx][bug] set_color no longer working

Fabrizio-Caruso, 2019-10-09:
@suborb `set_color` is also broken
```
	set_mode(mode_0);
	set_color(15, 1, 1);	
```
(compiled without msxbios) should set text mode 32x24 and set colors but it does not set any color.

Fabrizio-Caruso, 2019-10-10:
@suborb I can confirm that this issue is present today (2019/10/10).

suborb, 2019-10-10:
All `clrscr()` does is to emit character 12. It's up to the console driver to interpret this and do something.

Looking at the vt100 terminal, this is covered here:

https://github.com/z88dk/z88dk/blob/e08f8ce20bffcb161ec99e17d10f5a68723e3796/libsrc/stdio/ansi/f_ansi.asm#L73

And at generic console, this is covered here:

https://github.com/z88dk/z88dk/blob/e08f8ce20bffcb161ec99e17d10f5a68723e3796/libsrc/stdio/fputc_cons_generic.asm#L399

However, if I understand correctly you're using text 32x24 so you're bypassing the VT100 terminal (since that only works in graphics mode), and I have feeling you're not using the generic console.

So we're ending up in the MSX native `fputc_cons()` here: 

https://github.com/z88dk/z88dk/blob/e08f8ce20bffcb161ec99e17d10f5a68723e3796/libsrc/stdio/msx/fputc_cons.asm#L49

Which calls the firmware CLS routine. This routine will depend on the "mode" system variable which will only be set if you compile with `-lmsxbios`

Fabrizio-Caruso, 2019-10-10:
I have tried to use msxbios but it remaps the VDP and I am not sure how.
Should I assume it uses standard BIOS MSX mapping, i.e.
```
0000 | 07FF | VRAM: Main Tile Patterns (1/3)
0800 | 0FFF | VRAM: Extra Tile Patterns (2/3)
1000 | 17FF | VRAM: Extra Tile Patterns (3/3)
1800 | 1AFF | VRAM: Tilemap
1B00 | 1B7F | VRAM: Sprite Attributes
1B80 | 1BAF | VRAM: Palette Table
2000 | 37FF | VRAM: Colormap
3800 | 3FFF | VRAM: Sprite Patterns
```

Fabrizio-Caruso, 2019-10-10:
It works!
@suborb, @zx70 I confirm that msxbios does make everything work once you know how the VDP is mapped.

Fabrizio-Caruso, 2019-10-10:
@suborb, @zx70 what is not working is only the non `msxbios` case, which does provide `set_color` and `clrscr` functions but they have no visible effect and they may be thrashing (video-)memory.

suborb, 2019-10-11:
`msx_set_color()` has been fixed in the non msxbios case.

Given that the native `fputc_cons()` calls the firmware and that `clrscr()` calls `fputc_cons()` then unless you compile with `--clib=ansi` or `--generic-console` then you **MUST** use `-lmsxbios`

If `msx_set_color()` isn't correct then please reopen!

closed 2019-10-11 by suborb