z88dk/z88dk #1298: [cgetc][bug?] cgetc is supposed to wait for intput (getch?)

Fabrizio-Caruso, 2019-10-10:
cgetc() in CC65 is supposed to wait for input
https://www.cc65.org/doc/funcref-76.html

Is the current definition consistent with this behavior?

Maybe it should be defined as getch()

zx70, 2019-10-10:
I take it as a hint.  Thankyou

Fabrizio-Caruso, 2019-10-10:
@suborb  I thought I had seen `getk` with the same definition. `getk` does not wait for input. Does it?

suborb, 2019-10-10:
See here:

https://github.com/z88dk/z88dk/blob/e08f8ce20bffcb161ec99e17d10f5a68723e3796/include/conio.h#L127

Correct, `getk()` just polls and returns.

Fabrizio-Caruso, 2019-10-10:
@suborb then the current definition of cgetc is wrong because it is not doing what it does in other conio implementations including CC65

suborb, 2019-10-10:
The definition is correct, the implementation of `fgetc_cons()` on a particular target may be incorrect. Any particular target causing problems?

Fabrizio-Caruso, 2019-10-10:
cgetc is supposed to wait
getk is supposed not to wait
If both are defined as fgetc_cons, one of the two is wrong and getk is working as expected so the problem is cgetc

suborb, 2019-10-10:
I can't find anything in the tree that defines `fgetc_cons()` to `getk()` - can you give me a hint as to which target is causing problems?

The only thing I can find that could explain the behaviour you're reporting in the combined z88dk/CrossChase code base is this:

https://github.com/Fabrizio-Caruso/CROSS-CHASE/blob/5218cc2eff53c529d128f191f4eb83569014082c/src/cross_lib/input/input_macros.c#L248

zx70, 2019-10-10:
Fabrizio, cgetc is NOT mapped to getk..I did it yesterday and testing on zx spectrum worked as expected. 
It can be that on a particular target fgetc_cons is not working correctly,  which is the test where you saw problems? On which computer or emulator did you run it?

Fabrizio-Caruso, 2019-10-10:
The new definition causes problems in Cross Chase (tested in ZX81 version):
```
#if defined(Z88DK)
    #undef cgetc
    #define cgetc() getch()
#endif
```

zx70, 2019-10-10:
now it is easy, I'm changing the declaration in conio accordingly (with getch)

zx70, 2019-10-10:
A little bit of explanation,  we already faced similar problems and after a while we decided to create specific functions to emulate the console behaviour of other compilers / libraries (BDS C, Turbo C,, etc.. ). 
After a while the macros were substituted by a full emulation function which finally @suborb integrated in his rework of the console support:
https://github.com/z88dk/z88dk/blob/master/libsrc/stdio/kbhit.asm
it should be available tomorrow as usual

closed 2019-10-11 by zx70