z88dk/z88dk #1301: [zx81][getk] key 32 (space) detected instead of other key

Fabrizio-Caruso, 2019-10-12:
`getk` in Z81 target detects character 32 (space) when other keys are pressed.

You can reproduce this by running this program and keeping two keys pressed.

```
#include <conio.h>
int main()
{
    while(1)
    {
        printf("ch: %d\n", getk());
    }
    return 0;
}
```

Compiled by:
```
getk_test:
	$(Z88DK_PATH)$(MYZ88DK) +zx81 \
	-compiler=sdcc \
	-create-app -o  $(BUILD_PATH)/getk_test.prg \
	$(SOURCE_PATH)/../test/getk_test.c
```

zx70, 2019-10-12:
I'm guessing a possible fix, please check it tomorrow morning after the nightly build. 
It is based on the analisys of the ROM code,  but I couldn't test it.

suborb, 2019-10-12:
@zx70 it didn't work! I've fixed it up based on what the ROM does so now a double key press results in 0 being returned.

closed 2019-10-12 by suborb

zx70, 2019-10-13:
Dom, your ability to proceed in multitasking is impressive  8|

Fabrizio-Caruso, 2019-10-13:
@suborb you are superhuman! Thanks!
I have verified the fix! It does work!

commit 9422377f (2019-10-12): Fix #1301: Double keypresses on zx81/lambda resulted in wrong result

Space was always returned (since the code was 0). Follow the model
of the ROM to deal with this.