z88dk/z88dk #1302: sccz80 - enum compile error

feilipu, 2019-10-13:
I've been working on some code for an I2C LCD screen, which although still in development compiles fine with zsdcc. But, it doesn't compile with sccz80.
There is an issue with the `enum` definitions, it seems.

```bash
:~/Z80/src/i2c_test_c$ zcc +yaz180 -subtype=app -clib=new -v -m -SO3 --list test.c -o test

PROCESSING test.c
z88dk-ucpp -iquote"." -D__Z88DK -D__YAZ180 -D__Z180 -D__SCCZ80 -isystem"/home/phillip/Z80/z88dk/lib/config/../..//include/_DEVELOPMENT/sccz80"  -DSCCZ80 -DSMALL_C -D__SCCZ80  "test.c" "/tmp/tmpXXP0KWdB.i2"
z88dk-zpragma -sccz80 < "/tmp/tmpXXP0KWdB.i2" > "/tmp/tmpXXP0KWdB.i"
sccz80  -ext=opt -mz180 -standard-escape-chars   "/tmp/tmpXXP0KWdB.i" -o "/tmp/tmpXXP0KWdB.opt"
sccz80:"test.h" L:45 Error:A parameter named backLightSwi has already been defined for function
sccz80:"test.h" L:45 Error:Missing token, expecting ) got e
sccz80:"test.h" L:45 Error:Cannot initialise function 'LCD_WorkingModeConf' to a constant
sccz80:"test.h" L:45 Error:Missing token, expecting ( got m
sccz80:"test.h" L:45 Error:Definition of 'LCD_WorkingMode': int LCD_WorkingMode(int mode) - does not match declaration at "test.h":38 : int 
sccz80:"test.c" L:39 Error:Unknown symbol: LCD_ADDRESS
sccz80:"test.c" L:42 Error:A parameter named backLightSwi has already been defined for function
sccz80:"test.c" L:42 Error:Missing token, expecting ) got e
sccz80:"test.c" L:42 Error:Cannot initialise function 'LCD_WorkingModeConf' to a constant
sccz80:"test.c" L:42 Error:Missing token, expecting ( got m

Maximum (10) number of errors reached, aborting!
Compilation aborted
:~/Z80/src/i2c_test_c$ 
```
I've attached three files relevant, the `i2c.h` file (from z88dk), the `test.h` and the `test.c` files. All renamed to .txt because github.
[test.c](https://github.com/z88dk/z88dk/files/3721711/test.c.txt)
[test.h](https://github.com/z88dk/z88dk/files/3721712/test.h.txt)
[i2c.h](https://github.com/z88dk/z88dk/files/3721713/i2c.h.txt)

Note some odd code in the test.c, trying to work around the issue.
Is it me?

suborb, 2019-10-13:
I have a hunch. Hopefully I’ll get a fix for you by tomorrow.

feilipu, 2019-10-14:
Fixed, with thanks.

closed 2019-10-14 by feilipu