z88dk/z88dk #1303: (classic) Re-incorporate s1sdk

suborb, 2019-10-13:
The s1sdk was forked off z88dk years ago and died pretty much as soon as the rockchip players came in. The original site s1mp3.org has been down for a few years.

It's pretty hard to find these days and is obviously based on an ancient z88dk so I think it would be neat to bring it into the modern z88dk and provide a choice of compilers etc

I don't think I have any working hardware (I used to have an mp4 player I think) so this will have to be another target like the Rex where we can't actually test it.

zx70, 2019-10-14:
This one is challenging,  I think that little space was left for the user program, and only by replacing the program slot in EEPROM used by the OS for the radio application.
They were trying to create a full operating system.

suborb, 2019-10-14:
Yes, it's definitely a challenge, I'll settle for working on the emulator to be honest and this stage - I've not got access to any devices or any firmware files so verifying the firmware stuffing would be tricky.

There's a real dearth of technical information on archive.org - the wiki doesn't even have a memory map or list of ports and my copy of the mailing list archives are devoid as well.

The Swan thing was a bit strange - I don't think there were any developers behind it so it didn't move beyond a few paper proposals.

suborb, 2019-10-14:
It's a good bit of software archaeology, I've now found the correct start page and...

We can load into RAM over USB:

* https://web.archive.org/web/20081216060025/http://wiki.s1mp3.org/Loadram

From there there's a link to the loadram source code.

* Memory map: https://web.archive.org/web/20090210184143/http://wiki.s1mp3.org/Memory_map
* Port map: https://web.archive.org/web/20090213032359/http://wiki.s1mp3.org/Port_map
* Display driver: https://web.archive.org/web/20090213030111/http://wiki.s1mp3.org/Display

suborb, 2019-10-15:
After a fair bit of pain with the emulator, I've now got something working:

[image]

There's a couple of issues with the display:

1. It's upside down (this will be a library problem - other test programs have text the right way up)
2. "Page 0" of the display is at the bottom of the display - it should be at the top (so an emulator issue)

suborb, 2019-10-20:
I've merged it in #1307 - it needs a fair bit of work to be properly usable but it at least functions now.

closed 2019-10-20 by suborb