z88dk/z88dk #1305: sccz80 behavior changed in the last 20 days ?

zx70, 2019-10-15:
Sorry for the odd question but I think I could build and run the following program, almost with no changes.

https://github.com/z88dk/z88dk-ext/blob/master/misc/pi.c

I tried it again today and it gives obvious errors..
Was it me not noticing that I was building from a different source, or something changed in the last 20 days ?

suborb, 2019-10-15:
I think you need `-DPASCAL -DFAR` to get it compile

suborb, 2019-10-15:
So we need the following:

```
#define FAR
#define PASCAL
#define Size_T size_t
#define mlloc(x,y) calloc(x,y)
```

I think you must've used a different source with those corrections in it?

zx70, 2019-10-15:
Ok,  it is surely some stupid mistake I've done. 
Tnx

closed 2019-10-15 by zx70