z88dk/z88dk #1308: newlib - native timer implementation & time.h

feilipu, 2019-10-20:
I'm interested to hook a time of day function, using the z180 PRT, into a standard delay functionality. I can't see anything in the libraries (except the tm* to msdos* functions). Rather than building a specific z180 solution, it would be good to have a framework for all in time.

I have some pieces in place, [for the PRT](https://github.com/feilipu/yaz180/blob/master/yabios/asm_common1_driver.asm#L831) and for the [standard C time library](https://github.com/feilipu/z88dk-libraries/tree/master/time), but nothing for second or sub-second timing.

Perhaps an implementation of the `native_timer_start(void)` and  `native_timer_stop(void)` function would be a start. I'm looking to know time differentials, in fractions of a second, rather than day of week, calendar functions.

Thoughts and advice, please?

suborb, 2019-10-21:
Looking around classic, it looks like there's `clock()` implementation for a few targets which can be used to measure "elapsed" time. I'm not overly convinced by some of the implementations (ah, the benefit of hindsight!) and I don't think it's exactly what you want.

For delays, classic uses the busy looping newly function `asm_z80_delay_ms()` aliased to `msleep()` which I think breaks/overflows when clock speed > 6MHz. Given that it's busy waiting it's not exactly battery friendly.

In terms of timers, I can't see anything built in anywhere, there's the tick handlers for interrupts but no infrastructure built on top. So whatever you decide on I suspect I'll be lifting and incorporating into classic.

The "obvious" API to implement would be the POSIX timer API though whether it's suitable is another matter entirely.

feilipu, 2019-10-22:
Built enough. Might come back to it sometime to implement support for multiple clocks per device.

closed 2019-10-22 by feilipu