z88dk/z88dk #1312: [travis] svn at sourceforge is not responding

feilipu, 2019-10-27:
Hmm. Been dead for 3 hours now. 

```
svn checkout -r 11311 svn://svn.code.sf.net/p/sdcc/code/trunk/sdcc -q /tmp/sdcc

2094svn: E210002: Network connection closed unexpectedly

2095Makefile:71: recipe for target 'bin/zsdcc' failed

2096make: *** [bin/zsdcc] Error 1

2097The command "./build.sh -e -t" exited with 2.
```

feilipu, 2019-10-29:
Still dead. Two days now.

A sourceforge configuration change? 
Surely not an outage for more than two days?

suborb, 2019-10-29:
There was a ddos attack a couple of days ago. 

That stage has been quite failure prone in the past - maybe it should be removed from travis/nightly stage - for nightly we can use a pre-built version - I think it’s used in the z80asm tests.

suborb, 2019-10-29:
I've switched the url from svn:// to https:// so we're past that hump.

Now to see if anything else has broken!

closed 2019-10-29 by suborb

feilipu, 2019-10-30:
Out of interest, I just restarted the old [build using svn](https://travis-ci.org/z88dk/z88dk/builds/603453525), and it is working now.