z88dk/z88dk #1316: Question: GameBoy timings

pauloscustodio, 2019-11-01:
There is inconsistent information about gameboy instruction timings, in particular call, ret, jump and restart, in the references I found:
[1] https://www.pastraiser.com/cpu/gameboy/gameboy_opcodes.html
[2] http://marc.rawer.de/Gameboy/Docs/GBCPUman.pdf

call: 24 T in [1], 12 T in [2]
call cc: 12/24 T in [1], 12 T in [2]
ret: 16 T in [1], 8 T in [2]
ret cc: 8/20 T in [1], 8 T in [2]
rst: 16 T in [1], 32 T in [2]

Any idea what is correct?

suborb, 2019-11-01:
There's a resource here: http://slack.net/~ant/old/gb-tests/

That has call cc at 12T (always), looking around some other emulators they seem to agree. Given that consensus I'd suggest the first link is incorrect (which is of course where I got the timings from in the first place!)

pauloscustodio, 2019-11-01:
Ok. I will remove document [1] from techdocs and fix z80asm and ticks timings to use document [2].

I'll submit a PR for ticks.

commit c858c7ad (2019-11-01): Fix #1316: Ticks: fix gbz80 timings

STOP (0x10, 0x00) emulation: stop the simulation.
Emulate gbz80 I/O by reading/writing to memory 0xFF00+port.
Issue error on invalid gbz80 instructions.
commit 0ed8e765 (2019-11-01): Fix #1316: Ticks: fix gbz80 timings

STOP (0x10, 0x00) emulation: stop the simulation.
Emulate gbz80 I/O by reading/writing to memory 0xFF00+port.
Issue error on invalid gbz80 instructions.
closed 2019-11-01 by pauloscustodio