z88dk/z88dk #1318: Help: Travis-CI failing test/suites/string

pauloscustodio, 2019-11-02:
Help needed.
Travis-CI fails the test/suites/string on the feature/z80asm_pseudo_opcodes branch, but in my box both the msys2 and the Ubuntu builds pass the tests.

The Travis-CI build aborts with the message "The job exceeded the maximum log length" after an infinite sequence of:
```
zcc +test -vn -clib=8080  -I../../framework   -DNO_LOG_RUNNING -DNO_LOG_PASSED ../../framework/test.c strstr.c strcmp.c strncmp.c main.c strrstr.c strrev.c strlcpy.c strlcat.c stricmp.c -o test_8080.bin -lndos  -m
12:49:56  Starting 8080 test
z88dk-ticks -w 30 -m8080 test_8080.bin
Starting suite %s (%d tests)
%d run, %d passed, %d failed
Starting suite %s (%d tests)
Starting suite %s (%d tests)
%d run, %d passed, %d failed
Starting suite %s (%d tests)
Starting suite %s (%d tests)
%d run, %d passed, %d failed
```

Any ideas of what might be causing this?

suborb, 2019-11-02:
I’ve seen odd behaviour like this when jr emulation has been wrong - I don’t know if you tweaked it in that branch but that might be a good place to start.

pauloscustodio, 2019-11-03:
git bisect to the rescue!

The failure is caused by commit caa5deb4c4dd9476aa3bca1897b54ee6e4fb6e5a that makes z80asm accept the standard Intel opcodes when building for 8080 or 8085 machines.

The problem is:
- ```jp``` is jump-if-positive in Intel but unconditional jump in all other CPUs; Intel requires ```jmp``` for unconditional jump;

- ```cp``` is call-if-positive in Intel but compare in all other CPUs; Intel requires ```cmp``` for compare.

In library code written to be compiled in all CPUs, this causes problems as ```jp``` and ```cp``` assemble the wrong instructions when building for Intel.

Possible solutions, none of them of my liking:

1. Change the library code with ```IF __CPU_INTEL__``` to use the correct jump and compare instructions depending on the CPU; clobbers the code;

2. Change the library code to use the Intel ```jmp``` and ```cmp``` opcodes, which are accepted by z80asm in all CPUs; non-standard code;

3. Drop the acceptance of the ```jp``` and ```cp``` Intel opcodes in z80asm; impossible to assemble original Intel code;

4. Issue a warning on ```jp``` and ```cp``` when assembling for Intel; causes false positives;

5. Separate completely the Intel and Zilog opcode spaces, i.e. do not accept Intel opcodes when assembling for Zilog and vice-versa; extreme case of 1. above, forces rewrite of the library for Intel;

Any other suggestions? Which of these has the lesser impact?

suborb, 2019-11-03:
Rewriting the library in intel syntax is a non starter!

How about an option to only parse intel opcodes? By default Zilog would be accepted for the intel cpus. I guess it’s improbable that a file would contain both syntaxes?

closed 2022-05-11 by feilipu