z88dk/z88dk #1323: Submit docker image to Dockerhub

benjymous, 2019-11-08:
Any chance you could submit the docker image to dockerhub, so that there'll always be an up to date image available?

(It just needs someone with access to the github repo to create a dockerhub account, and link it to the github repo, and dockerhub should do the rest automatically)

feilipu, 2019-11-11:
Just checked, and the username _z88dk_ has been taken.
Not sure who has it, or whether another user name is acceptable.

suborb, 2019-11-11:
That was me registering the username, unfortunately the activation email bounced so now I can't get in.

I've opened a ticket with docker support so we'll see how it goes.

suborb, 2019-12-12:
And it's been a month, I've sent a chasing support email.

suborb, 2020-02-01:
There's now a docker build linked to master.

However, it takes 53(!) minutes to build so I think I'll investigate webhooks and bind it to the nightly build instead.

closed 2020-02-02 by suborb